artisan-roaster-scope/artisan #983: Feature request: color in properties

s4reea, 2022-10-13:
Hi
Few simple things
- I know a 0.x is not that much taste relevant. 
but my meter has this resolution and I always find myself typing in notes the exact color 80.7 for example. While in properties 81

- there isnt agtron gourmet
- if you add another field to state grinder settings used for color measurement, will help. Like avg size in uM

105.4 agtron gourmet at 100um

MAKOMO, 2022-10-13:
Being a producer of a color meter device myself I can ensure you that decimals are not relevant here. Not influencing the taste nor is there any affordable instrument seriously being able to asses color with that precision and accuracy. Take another copy of the same instrument, recalibrate your instrument, measure in a room with a slightly  different temperature, re-prepare your sample,.. you will not get the same reading up to the decimal. Decimals are misleading and make the results harder to read and interpret.

There is ONLY the Agtron Gourmet entry abbreviated just as Agtron as nobody seriously uses the Agtron Commercial scale those days.

Regarding grind size is relevant, but different devices have different requirements. The particle distribution is important, not the average particle size and it should be kept constant to get comparable results. However, how to relate two color readings done at different average particle size? Totally unclear. Therefore the indication of the average particle size is not productive.

MAKOMO, 2022-10-13:
I added a change such you can now type color values as "82.7" or "82,7" but those values will be rounded to the next Integer.

s4reea, 2022-10-13:
Thank you very much I appreciate your effort
I agree with particle size distribution, when I recalibrated my burrs my color axis changed a lot.

closed 2022-11-30 by MAKOMO