artisan-roaster-scope/artisan #1001: Artisan 2.8.0 only running in root mode in ubuntu 22.04

cyl0tera, 2022-11-05:
Hi all,

In advance sorry for my bad english.
I have installed new version artisan 2.8.0 on ubuntu 22.04

When i want to run artisan by clicking on the shortcut, nothing happen.
I have to run artisan with terminal " sudo artisan" and then it's working.

Have you got any solution please ? 

 - Artisan Version 2.8.0
 - Artisan Build 49b6c9c
 - Ubuntu 22.04.1 LTS
 
Thanks for your help

MAKOMO, 2022-11-05:
Works for me. How did you install Artisan? What happens if you start it (without sudo) like this

```
# /usr/share/artisan/artisan
```

Do you see any error output?

cyl0tera, 2022-11-05:
I install .deb by moving it on firefox and who open the native ubuntu packets installator

/usr/share/artisan/artisan: symbol lookup error: /lib/x86_64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_proxy_marshal_flags

philiptbuckley, 2022-11-06:
Same issue here on Ubuntu 22.04 & 22.10 (I tried upgrading Ubuntu but same problem). 
Both of these ubuntu versions run Wayland display server by default. If I log in with xorg (log out and click gear icon before entering your password, select xorg) it works fine.
As cyl0tera states -- also works as root (with warning QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root').
Artisan 2.4.6 doesn't suffer this issue

Googling _undefined symbol wl-proxy-marshal-flags_ shows that other applications also suffer this problem

cyl0tera, 2022-11-06:
I just tried logging in with xorg and I don't have the problem it works perfectly

lookingafar, 2022-11-15:
[image]
I can confirm for Manjaro 22.0.0 (debian package installed from AUR)
If I run Artisan with Wayland session as a regular user, Artisan starts, but the title bar is missing for all artisan windows (see the screenshot). If I start Artisan as root in Wayland, the top window title bar appears. It also appears if run with xorg as normal user.

MAKOMO, 2022-11-23:
The only change between v2.6.0 and v2.8.0 that seems to be relevant here is the update of the GUI lib Qt from v6.2.3 to v6.4.0. Maybe Qt 6.4.0 introduced this issue? Does this problem also occur with the current v2.8.1 continuous build based on Qt 6.4.1 available at https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous?

ereyes01, 2023-02-23:
+1 I hit the same problem in Ubuntu 22.04 with artisan 2.8.2. Workaround of switching to Xorg unblocks me as well.

spwoolet, 2023-04-17:
I found that by modifying the artisan.desktop file and using
Exec=pkexec --user steve env DISPLAY=:0 XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.1U4A31 artisan
I am able to run as non-root and any created files are saved with appropriate permissions. Although I am required to enter my password to start the program.
The values for DISPLAY and XAUTHORITY were obtained by running
"echo DISPLAY=$DISPLAY XAUTHORITY=$XAUTHORITY" at a terminal prompt.

MAKOMO, 2023-05-05:
I have hopefully fixed the issue with Wayland and also a regression recently introduced by upgrading to Qt6.5 which caused a failure on startup under XOrg (Commit c6e2556bda7f84634be2c49fb7be59abef41b5cf).  

I have tested the current continuous builds

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

under Ubuntu 23.04 and Artisan started fine under both, Wayland and XOrg. Could you please verify this?

If you find any further issues, please report them here!

lookingafar, 2023-05-08:
On Ubuntu 22.04 on Virtualbox the continuous build started without problems as a normal user on Wayland (before the fix it only started as root on the same virtual machine)

MAKOMO, 2023-05-09:
Thanks for this confirmation!

closed 2023-05-09 by MAKOMO