artisan-roaster-scope/artisan #1002: Artisan 2.8.0 problem with Probat P12/2

grebdnul, 2022-11-06:
**Describe the bug**
Modbus TCP connection appears to stop working after a few gas or air changes has been sent to roaster. Communication works fine with Artisan version 2.6.0. 

**To Reproduce**
Steps to reproduce the behavior:

Select "Propat P-series 7" as machine and add air slider.
Start and send a few airspeed-changes or gas changes. 

**Expected behavior**
Expected behavior is that the roaster reacts to every gas and air-change.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 
 - Version [e.g. 22]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: mac OS 11.7.1. 
 - Connected devices or roasting machine [e.g. Probatone 5]: Probatone P12/2

**Additional context**
Problem started on MacOS 12 with direct TP cable to roaster.
Have downgraded to MacOS 11.7.1 and also added router with calble connected fixed ip-adresses for computer and roaster. 
Same problem on both MacOS 12 and 11 with Artisan 2.8.0. Uninstalled and installed Artisan 2.6.0 and everything works fine. Could it have something to do with new TCP-settings in 2.8.0?

[artisan-settings_P12_nyast.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/9946728/artisan-settings_P12_nyast.aset.zip)

(Thank you for a great roasting software.
regards, Philip Lundberg, Sweden)

MAKOMO, 2022-11-06:
Philip, your settings look ok. Could you please send me your Artisan log file to check if I can read there what is going on here? Just click the plus icon in the left top corner while holding the OPTION key pressed. This will open an email with those log files attached. Please send them to info@artisan.plus.

MAKOMO, 2022-11-23:
The issue could be duplicated and is related to the optimizer in the MODBUS and S7 module in Artisan not covering "external" reads as required to send keep-alive pings on controlling Probat/K&M machines.  Commit 6e92f28880b33796cf1b027c46f9d7e5ae7a953c fixes this.

closed 2022-11-23 by MAKOMO