artisan-roaster-scope/artisan #1012: Artisan Software shutting off mid roast

tmarshall2022, 2022-11-20:
The Artisan software has been shutting off completely mid roast.

It started just before 2.6.0 upgrade and continues in 2.8.0. 
Tried rebooting numerous times after each software crash, reloaded 2.6.0 software, reseating widget usb a couple times. 
Software crashes randomly, sometimes mid roast, late roast and does not seem temp specific. 

I’m at a loss with how to fix this and any advice would be appreciated.

bigl00z3, 2022-11-20:
You will need to give more details on your setup, what type of machine, which devices are connected, computer OS... if you want people to be able to help you. Just mentionning it crashes is not enough info.

The bug report templates actually gives you a good recomendation on what to provide when documenting properly an issue.

Also attach your artisan settings to your post `Help>>Save Settings...`   and then rename the .alog file to a .txt file before uploading

You can also look for the logs to see if there is a specific message or error related to the crash:
An OPTION click to the plus icon opens an email message with the log files attached.

MAKOMO, 2022-11-20:
Please share your Artisan settings as Nico explained as well as your log file. Both files need to be renamed to end in .txt and drag-and-dropped here on Github in this thread to a message box. Thanks for your support!

tmarshall2022, 2022-11-20:
Thank you for reaching out and please see additional info as requested:
Diedrich IR 2.5kg Roaster attached to a Dell PC with a Windows 10 operating system.
I've attached the Artisan Settings below in a text file. 

Unfortunately, there are no individual log files for the affected roasts. When the software crashed as I would reopen Artisan and  it would re-initiate as if there was no roast. Perhaps I'm missing something in your request regarding attaching log files. Hope this provides some additional information and thanks in advance. 

[Artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/10050616/Artisan-settings.txt)

MAKOMO, 2022-11-20:
Thanks for attaching your settings, but we need also your Artisan log file:

```
You can also look for the logs to see if there is a specific message or error related to the crash:
An ALT click to the plus icon opens an email message with the log files attached.
```

Rename also this log file adding the .txt extension to artisan.log.txt and attach it here on Github via drag-and-drop to a message window too.

Thanks!

tmarshall2022, 2022-11-20:
Thanks for trying to help. The files that crashed weren't saved, so I don't have those files to send to you. Is there anything else that I can attach that would be helpful? I did 10 roasts yesterday. The 6 that saved, worked fine. The 4 that didn't save were the one's that crashed.

MAKOMO, 2022-11-20:
I am asking for the technical log (NOT the Artisan profiles; those got lost, I understand this) file that Artisan writes. Just click on the plus icon in the upper left corner of the main Artisan window while holding the ALT key as I asked you for. The email message that will open has attached that technical log file. Please rename this with .txt extension and attach it here.

MAKOMO, 2022-11-24:
Besides having the log file for a check, it would be good to know if you are running the latest Phidget driver and firmware on your 1048. Please install the lastet driver from the Phidgets home page and then open its Phidget Control Panel to check if the 1048 runs on the latest firmware (not marked in red).

tmarshall2022, 2022-11-24:
Hi Marko. Thanks for reaching out. I’ve been struggling with the Microsoft Outlook prompt with the Alt + to generate an activity log. No sure how I am to generate a Microsoft exchange server number as our ISP does not provide this so I’m a little stuck. 
I have someone who call help me generate the activity log if you could articulate the pathway to the activity log?
I will update the Phidget Driver as you suggested as well. 
Thank you and my apologies for the extra efforts.  
Trent

roasterdave, 2022-11-24:
Don't wrestle with Outlook.  You can grab the log files directly at:
C:\Users\<youraccount>\AppData\Local\Artisan-Scope\Artisan\artisan.log
C:\Users\<youraccount>\AppData\Local\Artisan-Scope\Artisan\artisan_device.log

MAKOMO, 2022-11-30:
Any news on this?

tmarshall2022, 2022-11-30:
Hi Marko. I updated the Phidgets and will try again this weekend. Alternatively, I will upload the software to another computer and try that as well. Will advise how this works. 
Trent

tmarshall2022, 2022-12-03:
Morning Marko. I updated the Phidgets and did 9 roasts without problems or crashing. I will roast again next week and update you so you are aware of the potential resolution. Thank you for your help. Trent

closed 2022-12-15 by MAKOMO