artisan-roaster-scope/artisan #1016: i have installed artisan v2.4.2 in my Mac, but dosen't recognize phidgets temp in the program, all ready closed the Phidget app and docent work

AlejoTuesta, 2022-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:
 - Version [e.g. 22]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:
 - Connected devices or roasting machine [e.g. Probatone 5]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2022-11-25:
Why are you using an outdated Artisan version? Does the Phidget Control Panel show the temperatures from your modules?

AlejoTuesta, 2022-11-25:
i Have a Mac mini with a lowest version of OS and th new artisan requires a new version,  th problem I hv is tt. artisan doesn't get information from the phidget but the phidget has data in the Mac, here some images:
[image]

[image]
[image]

MAKOMO, 2022-11-26:
The [latest Artisan](https://github.com/artisan-roaster-scope/artisan/releases/latest) comes with a legacy build for macOS which runs on 10.13. Just download and install [```artisan-mac-legacy-2.8.0.dmg```](https://github.com/artisan-roaster-scope/artisan/releases/download/v2.8.0/artisan-mac-legacy-2.8.0.dmg). 

From your screen shot it seems you are not running the latest Phidget driver. Download and install the latest driver tagged ```OS X 10.11+``` from the [Phidgets download page for macOS](https://www.phidgets.com/docs/OS_-_macOS).

In Artisan under menu ```Config >> Device...``` make sure you have selected in the ```ET/BT``` tab under ```Meter``` the entry ```Phidget 1048 4xTC 01```.

MAKOMO, 2022-11-30:
Any success?

AlejoTuesta, 2022-11-30:
yes today I installed the artisan and works fine but I have download the latest Phidget driver and didn't work I had to reinstall the old version and funtion well, thanksFOR THE HELP

MAKOMO, 2022-11-30:
In which way did the current Phidget driver not work with your 1048? Which older version of the driver did you installed instead?

bigl00z3, 2022-11-30:
When updating the Phidget driver, the 1048 may also require a firmware upgrade to work correctly. 
This can be done via the Phidget Control Panel. You should see there if there is an update available for your device.

MAKOMO, 2022-12-15:
Stall

closed 2022-12-15 by MAKOMO