artisan-roaster-scope/artisan #1036: Modbus Problem

marufwahyup, 2022-12-13:
Modbus Communication Error while using Artisan 2.8.0 using my hardware Arduino mega with atmega16u2 serial, if I'm using the Arduino mega with usb2serial such as ch340 it'll be fine and I found an interesting problem that if I'm using the old version of artisan such as 1.1.0 they're all fine. Any suggestion for this problem? Thank you.

I'm using windows 10 
Hardware using both Arduino mega with atmega16u2 serial and Arduino mega with ch340 serial

MAKOMO, 2022-12-13:
Could you please test if the current continuous build performs any better: https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

This one comes with an updated MODBUS infrastructure.

Please report!

marufwahyup, 2022-12-13:
I'll give it a try. Thank you

marufwahyup, 2022-12-13:
It's still the same, but it's different artisan switching between connected to modbus and modbus disconnected. I've tried using my second arduino mega with ch340g it's fine connected via modbus.

MAKOMO, 2022-12-13:
Could you please activate the debug modus by clicking on the plus icon in the left upper corner while holding the keys OPTION+COMMAND (macOS) or ALT-CTRL (Windows) and also tick the "Logging" flag at the bottom of the Devices dialog (menu Config >> Devices).

Then please test the connection and after the test click again the plus icon, but just hold the OPTION (macOS) or ALT (Windows) key. Now an email should open with the log file attached. Please attach the log file here and I will analyze it. Thanks for your support!

marufwahyup, 2022-12-13:
Alright I'll try it

marufwahyup, 2022-12-13:
[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/10221323/artisan.log)
[artisanViewer.log](https://github.com/artisan-roaster-scope/artisan/files/10221325/artisanViewer.log)

There it is

MAKOMO, 2022-12-13:
I don't see any messages from the pymodbus lib in this log. Are you sure the "Logging" flag in the Device dialog is ticked?

Otherwise this log only shows that Artisan could not at all connect to the device. It receives an IO Exception from the underlying pymodbus lib.

Did you install/update the serial2usb driver for the atmega16u2serial module?

There seems to be a device driver needed from Atmel, right?

https://support.arduino.cc/hc/en-us/articles/4408887452434-Flash-USB-to-serial-firmware-in-DFU-mode

marufwahyup, 2022-12-13:
I did tick the logging flag in the Device dialog. I already flash a new firmware in arduino for the atmega16u2 module. How the old version of software in this case 1.1.0 can read the bean and environment temperature? Or is there any version that used the same pymodbus lib with version 1.1.0 because I can't build it from source yet?

MAKOMO, 2022-12-13:
Artisan v1.1.0 was build (in 2017!!) on Python 2.7 which is unsupported software by now and used very old pyserial and pymodbus libs. I have no idea why that version works on your usb2serial module while newer Artisan versions do not (could be a timing issue or anything else). Very strange. There thousands of projects based on pyserial/pymodbus and there are no corresponding connect issues known for the current versions. This must be something specific on lower levels. Anyhow the Artisan project has no influence on this. I fear we cannot help here on the Artisan level. I will close this issue therefore.

closed 2022-12-13 by MAKOMO