artisan-roaster-scope/artisan #1039: Dark Mode for Artisan

KodakC41, 2022-12-14:
### Discussed in https://github.com/artisan-roaster-scope/artisan/discussions/1038

<div type='discussions-op-text'>

<sup>Originally posted by **KodakC41** December 14, 2022</sup>
# Dark Mode

I believe that Artisan would benefit from a dark mode for users who are roasting, cupping, or profiling in the evening. I would be happy to implement this. 

 -*C*</div>

MAKOMO, 2022-12-14:
Dark mode is supported on macOS. Are you on Windows or Linux? Any idea how Windows handles this? This has to be applied on the PyQt/Qt level. Not sure there is a corresponding API. Do you know any details on this?

KodakC41, 2022-12-14:
I was taking a look at the graph in MacOS and it seems that the application bar is supported to match the theme of the device (Dark/Light), but the rest of the application remains white. Is this something that needs to be configured in settings or should I look into the issue?

MAKOMO, 2022-12-14:
menu ```Config >> Themes```

closed 2022-12-14 by MAKOMO