artisan-roaster-scope/artisan #1051: PDF Export not working in Artisan v2.8.0

Brewzologist, 2022-12-18:
**Describe the bug**
Exporting a roast log to a PDF either through File/ConvertTo/PDF or via automated export at end of roast doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File/ConvertTo/PDF
2. Follow process to select a roast log and target folder
3. PDF is not created.

**Expected behavior**
PDF file is created from the selected roast log

**Screenshots**
N/A

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.0
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 682bdaf
 - Version [e.g. 22]: N/A
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10
 - Connected devices or roasting machine [e.g. Probatone 5]:  Phidgets and Arduino

**Additional context**
See errors in attached artisan log file.  I added the "======" line in the log file to indicate the error in ConvertTo/PDF
[PDF Error.zip](https://github.com/artisan-roaster-scope/artisan/files/10254277/PDF.Error.zip)

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2022-12-18:
Duplicate. See Issue #1011.

closed 2022-12-18 by MAKOMO

roasterdave, 2022-12-18:
Thanks Steve for the detailed report. We appreciate that.

Brewzologist, 2022-12-18:
Thanks guys. Sorry for the duplicate bug report. Happy holidays!