artisan-roaster-scope/artisan #1065: Aillio R1: unable to detach kernel driver

Alexmitran, 2022-12-26:
**Describe the bug**
After connecting the roaster to the Mac and loading the settings when you start and  press ON this error appears :
Aillio R1: unable to detach kernel driver and no reading appears

**To Reproduce**
Steps to reproduce the behavior:
Load Aillio Bullet IBTS settings 
Press ON

**Expected behavior**
Error on reading

**Screenshots**
[image]

The device appears but it can not connect

[image]

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: V2.8.2
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: f525b16
 - Version [e.g. 22]: 
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: macOs Ventura 13.0
 - Connected devices or roasting machine [e.g. Probatone 5]: Aillio R1 IBTS

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.
[aillio.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/10304440/aillio.aset.txt)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2022-12-26:
If you downgrade to Artisan v2.8.0, does Artisan connect then again to the R1? There was one change between those two Artisan version which is an upgrade of the libusb library. That could be the culprit.

Alexmitran, 2022-12-26:
I tried downgrading and i got the same error. Thank you.Sent from my iPhone

MAKOMO, 2022-12-26:
I was digging a little deeper. The error occurs where Artisan tries to deattach the USB driver to be able to configure the port. This seems to be not possible for non-root users any longer with the latest macOS versions (from 10.12 or 10.13 on). I changed the app signing/notarization to request the right for this and uploaded the app with the changed signature to my DropBox:

https://www.dropbox.com/s/wekvwnmylel2y3h/artisan-mac-2.8.2.dmg?dl=0

This most likely requires you to allow USB access for Artisan in the macOS System Settings under Privacy & Security; scrolling down to the Security section. There should be an entry asking for permission.

If this does not work out then I would need an R1 bullet to investigate further.

Alexmitran, 2022-12-27:
Hi Marko,

I downloaded the software but the OS classified it as corrupted and didn’t open it.
I send you a video in the link:
https://we.tl/t-0OwFTD8wgy￼
MailVideo.mov
we.tl
https://we.tl/t-0OwFTD8wgy
Also I also saved the report

￼

Thank you again.
P.S. If it will help I will be more then welcome to send you the roaster and also a Phidget Scale kit that we talked about in Milan.

> I was digging a little deeper. The error occurs where Artisan tries to deattach the USB driver to be able to configure the port. This seems to be not possible for non-root users any longer with the [...]

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Incident Identifier: 1F03D899-6A48-43BE-9C59-FFD239FE5201
CrashReporter Key:   A1E261E4-1888-A2D6-1B6E-9664E2A20AD8
Hardware Model:      MacBookPro14,2
Process:             Artisan [4821]
Path:                /Volumes/*/Artisan.app/Contents/MacOS/Artisan
Identifier:          org.artisan-scope.artisan
Version:             2.8.2 (Artisan 2.8.2)
Code Type:           X86-64 (Native)
Role:                Default
Parent Process:      launchd [1]
Coalition:           org.artisan-scope.artisan [1038]

Date/Time:           2022-12-27 13:44:39.4536 +0200
Launch Time:         2022-12-27 13:44:39.3806 +0200
OS Version:          macOS 13.1 (22C65)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGKILL (Code Signature Invalid))
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: CODESIGNING 1 Taskgated Invalid Signature

Triggered by Thread:  0

Thread 0 Crashed:
0   ???                           	       0x10449def0 _dyld_start + 0
1   ???                           	               0x1 ???

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x0000000000000000  rsi: 0x0000000000000000  rbp: 0x0000000000000000  rsp: 0x00007ff7bfeffbf0
   r8: 0x0000000000000000   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x000000010449def0  rfl: 0x0000000000000200  cr2: 0x0000000000000000
  
Logical CPU:     0
Error Code:      0x00000000 
Trap Number:     0

Binary Images:
       0x104499000 -        0x104530fff ??? (*) <bb7a0970-8c62-3dce-a7a2-5cec9c501f11> ???
               0x0 - 0xffffffffffffffff ??? (*) <00000000-0000-0000-0000-000000000000> ???

Error Formulating Crash Report:
dyld_process_snapshot_get_shared_cache failed

EOF

-----------
[... 77 lines cut ...]
      "deploymentId" : 240000023
    }
  ],
  "experiments" : [

  ]
},
  "reportNotes" : [
  "dyld_process_snapshot_get_shared_cache failed"
]
}

Model: MacBookPro14,2, BootROM 499.40.2.0.0, 2 processors, Dual-Core Intel Core i5, 3,1 GHz, 8 GB, SMC 2.44f6
Graphics: Intel Iris Plus Graphics 650, Intel Iris Plus Graphics 650, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x80CE, 0x4B3445364533303445422D45474347202020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x171), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1764)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB30Bus
USB Device: Apple T1 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.5

MAKOMO, 2022-12-27:
Hi, I am travelling now in Italy so I will not be able to investigate this right now further. I can duplicate your crash here too. I only added one entitlement to allow USB access and too the released build for signing, thus I did not expect this to crash and therefore didn't test it before sending. Sorry. So it seems that Artisan is for now not able to talk to USB devices on macOS 10.13 at all. No idea how to fix this. This is a permission issue by the OS, not an implementation issue. Have to think about it. Got my own Phidget Scale kit by now, but did not even tested it yet. Hope to find the time for this this spring.

MAKOMO, 2022-12-27:
Is your mac an M1 or M2 Apple silicon or an Intel-based machine?

Alexmitran, 2022-12-28:
It is intel base. I tested also on an M1 but didn’t connect and had the ‘kendal” error.I put a Virtual Machine and it work fine.If you don’t mind I have a question regarding the PID from Artisan. The artisan can control the Heater, can it be set to work for the PID in the Artisan  as the PV? So i can set a SV and the Artisan will modulate the heater ?Thank you and feel free to answer me anytime.Sent from my iPhone

MAKOMO, 2022-12-28:
So you tested on Intel and M1 mac under macOS Ventura and it failed on both with the same kernel detach error, right?

What exactly did you do to get it working anyhow? What do you mean exactly by "I put a Virtual Machine and it worked fine"?

Artisan contains an internal software PID which can observe a target (was only ET/BT before the current release, but now any device input can be a target) and operates any Artisan slider (most useful a slider controlling the heater of the machine) to reach the specified set value (SV). The SV can be manually set, be set by a Ramp/Soak profile or can also come from a background profile. 

I use the PID with a manual SV for pre-heating and my BBP on my gas-driven drum roaster.  On air roasters, or drum roasters that react fast on heater changes, one can successfully follow a background profile, also if "drawn" using the Artisan designer.

See my post on the PID for all the details:
https://artisan-roasterscope.blogspot.com/2016/11/pid-control.html

Juppi1252, 2023-01-06:
Hi,
I get the same error message. Same devices, same constellation. When downgrading to 2.8.0, the system works properly again.

MAKOMO, 2023-01-06:
Hm. This is strange. This would indicate that either the detach is not needed and/or there is an issue in the latest libusb.

I did comment this detach that raises this error and triggered a new continous build. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Could you please test if this works again for you? Note you need to start the app first time via a right-click to the app icon selecting Open as this build is not signed nor notarized.

If this does not work for you I can still try to downgrade libusb to the previous version.

Thanks for testing and reporting!

Juppi1252, 2023-01-06:
First Test: It works.
Thanks a lot
Josef

Juppi1252, 2023-05-13:
> kernel driver (Issue #1065)
>
>
>
>
> @Alexmitran <https://github.com/Alexmitran> : we've just uploaded a
> (signed) beta of the next Artisan release for review by Artisan experts
> like you.
> This version includes only a few minor new features, but lots of bug fixes.
> Release History:
> Maybe you find time to test this version on your setup and report back
> issues.
> Thanks,
> Marko & Dave
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "https://github.com/artisan-roaster-scope/artisan/issues/1065#issuecomment-1545530872",
> "url":
> "https://github.com/artisan-roaster-scope/artisan/issues/1065#issuecomment-1545530872",
> "name": "View Issue" }, "description": "View this Issue on GitHub",
> "https://github.com" } } ]

Hello Marco, you made version 2.8.3 available to me a few months ago and I 
tested it immediately, the message with the kernel driver no longer 
appeared. Everything is going fine.
 
LG
Josef

MAKOMO, 2023-05-14:
Thanks Josef for testing and this confirmation!