artisan-roaster-scope/artisan #1076: Artisan not opening on Windows 10 Pro

kosgeiz, 2023-01-16:
## Expected Behavior
We have downloaded Artisan from the official website (latest version)

## Actual Behavior
Initially Artisan was asking for python311.dll file which once we reinstalled it was able to locate the file.

Now artisan does not even open or show are error

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - Artisan Version: **artisan-win-2.8.2**
  - Artisan Build (number in brackets shown in the about box): 
  - Platform (Mac/Windows/Linux + OS version): **Windows**
  - Connected devices or roasting machine: **Besca**

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file. **File was not able to open**
Please attach any relevant Artisan *.alog profiles.  **File was not able to open**

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

roasterdave, 2023-01-16:
Please confirm Windows is not is S mode.  Type "system' into the Windows search box to see.  Was an earlier version of Artisan previously installed and running on this machine?  Did the same Windows user install and try to run Artisan?

From the description, the initial install was corrupt and likely the second one as well.  Please perform an uninstall by typing "remove programs" in the Windows search box and selecting Artisan or by running **c:\Program Files\uninst.exe**  After the uninstall completes, check that the folder **c:\Program Files\Artisan** was removed.  If it is still there, delete all the files within and then delete the folder.  Next reinstall Artisan.

poundy, 2023-01-23:
Artisan has worked reliably on my machines, both Win10 and Win11, for many years. This does not look to be an issue with Artisan itself but with your PC/configuration. As Dave mentioned, you will need to try uninstalling/reinstalling and making sure no remnants are left behind (which Artisan is good at doing, but if something has happened with your previous installs it may not have been able to achieve that on it's own). Please keep us posted.

roasterdave, 2023-02-16:
Stall

closed 2023-02-16 by roasterdave