artisan-roaster-scope/artisan #1078: Error using Apply Button - Events/Quantifiers

renatocipriani, 2023-01-19:
**Describe the bug**
This error message when trying to change "events/qwuantifiers" value

  File "artisanlib\events.py", line 1569, in applyQuantifiers   File "artisanlib\main.py", line 21845, in digitize  -------------------------------------------------------------------------------- artisanlib\events.py::applyQuantifiers:1569 	                self = <artisanlib.events.EventsDlg object at 0x000001D104711DA0> 	                   _ = False 	              redraw = False 	                   i = 0 	                temp = [70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0, 60.0, 50.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 40.0, 53.33333333, 66.66666667, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 80.0, 76.66666667, 73.33333333, 70.0, 70.0, 70.0, 70.0, 70.0, 70.0] 	               timex = [1.9791275, 4.9805404, 7.9802461, 10.9826219, 13.9822298, 16.9808552, 19.9784238, 22.9788395, 25.9778581, 28.9799227, 31.9789768, 34.9810467, 37.9798744, 40.980489, 43.9794865, 46.9806254, 49.9805314, 52.9808108, 55.9832486, 58.9796225, 61.9823378, 64.9791171, 67.9837741, 70.9800137, 73.9793318, 76.9843313, 79.9803918, 82.9838195, 85.978877, 88.9806846, 91.9802037, 94.9794971, 97.9813835, 100.9791518, 103.9815035, 106.9793074, 109.9788679, 112.9785883, 115.9774185, 118.9845767, 121.9802542, 124.9809191, 127.9832748, 130.9791191, 133.9781774, 136.9823901, 139.9780609, 142.9831276, 145.9815174, 148.9779389, 151.9818221, 154.9786841, 157.9782941, 160.9784449, 163.9804841, 166.9835621, 169.9821635, 172.9792951, 175.9830439, 178.9796153, 181.9787724, 184.9813809, 187.9781928, 190.9809899, 193.9810346, 196.9797991, 199.9821531, 202.9786546, 205.9810263, 208.9779076, 211.9801386, 214.9824151, 217.9790367, 220.978238, 223.9810852, 226.9842336, 229.9834804, 232.9828771, 235.9787954, 238.9809495, 241.979948, 244.9790104, 247.982613, 250.9792699, 253.9818362, 256.9788707, 259.9812917, 262.9836394, 265.9829129, 268.9816633, 271.9800467, 274.9833989, 277.9806671, 280.9790313, 283.9808988, 286.9796822, 289.9823669, 292.980893, 295.9830304, 298.9820095, 301.9789044, 304.982723, 307.9797807, 310.9794339, 313.9824891, 316.9789673, 319.9790848, 322.9830732, 325.9802904, 328.9798054, 331.9829132, 334.9788206, 337.9843294, 340.9805113, 343.979694, 346.979327, 349.9786068, 352.9828551, 355.9790792, 358.9796516, 361.9825973, 364.9779693, 367.9795236, 370.9787673, 373.9805467, 376.9818787, 379.9809567, 382.9802666, 385.9817887, 388.9785283, 391.9782185, 394.9843012, 397.9800841, 400.9789525, 403.9812547, 406.9808028, 409.9836352, 412.9788915, 415.9818739, 418.9808539, 421.9779971, 424.9802002, 427.980011, 430.9802259, 433.9795724, 437.0056715, 439.9820017, 442.9790883, 445.977844, 448.9828076, 451.9781426, 454.983589, 457.9806589, 460.980886, 463.9800853, 466.9789775, 469.9814858, 472.9809752, 475.9787229, 478.9813434, 481.9839815, 484.9804588, 487.9830535, 490.9787755, 493.9811753, 496.9813927, 499.9801214, 502.9781498, 505.9810251, 508.9780564, 511.982276, 514.977844, 517.9792353, 520.9834236, 523.9799657, 526.9787653, 529.9826963, 532.9785018, 535.9804918, 538.9836669, 541.9789834, 544.9777732, 547.9810105, 550.9802304, 553.980782, 556.9836467, 559.9786696, 562.9778504, 565.9841089, 568.9807655, 571.9832039, 574.9797037, 577.9796568, 580.9821278, 583.980778, 586.9828059, 589.9819628, 592.9780316, 595.9811027, 598.9783422, 601.9810072, 604.9837356, 607.9794988, 610.9836062, 613.9792797, 616.9787374, 619.9819572, 622.981008, 625.9780474, 628.9836563, 631.9788351, 634.9811427, 637.9803476, 640.9833142, 643.9828448, 646.9782537, 649.9801875, 652.9829863, 655.9786734, 658.9807303, 661.979077, 664.9776057, 667.9833883, 670.9791435, 673.9812117, 676.9833005, 679.9801965, 682.9831616, 685.9793313, 688.9794722, 691.981821, 694.9781435, 697.9806295, 700.9796473, 703.9798015, 706.9820792, 709.981174, 712.9817838, 715.9811602, 718.9787328, 721.9840369, 724.9795211, 727.982611, 730.9780943, 733.9804091, 736.9831972, 739.9790944, 742.9807991, 745.9773521, 748.9781271, 751.9809609, 754.9780795, 757.9811879, 760.9838712, 763.9801136, 766.97851, 769.9811974, 772.9783113, 775.9804289, 778.9823422, 781.9793321, 784.9807141, 787.9825903, 790.9786528, 793.9802556, 796.9811012, 799.980443, 802.9827318, 805.9784945, 808.9793126, 811.9808857, 814.9779625, 817.9837906, 820.9793693, 823.9785184, 826.9791087, 829.9784551, 832.9810803, 835.9779077, 838.9832102, 841.9802187, 844.9782879, 847.9810966, 850.9802106, 853.9786236, 856.9807996, 859.9802545, 862.980195, 865.9798246, 868.9825159, 871.979349, 874.9785462, 877.9789929, 880.9784525, 883.9836588, 886.9789605, 889.9819203, 892.9778154, 895.9803634, 898.9862516, 901.9838998, 904.9782869, 907.9835615, 910.9788084, 913.984208, 916.9834732, 919.9793898, 922.9827426, 925.9791042, 928.9780872, 931.9884822, 934.9807529, 937.9832938, 940.9793961, 943.9789083, 946.9824184, 949.9788887, 952.9791544, 955.9820622, 958.9804993, 961.9810342, 964.9804931, 967.9828841, 970.9805108, 973.9832728, 976.9803823, 979.9799289, 982.9831736, 985.9788693, 988.9786943, 991.9781276, 994.9802572, 997.983174, 1000.9791763, 1003.9824539, 1006.9791784, 1009.9828764, 1012.9786905, 1015.9789878, 1018.9806299, 1021.9805611, 1024.980846, 1027.9806705, 1030.9803124, 1033.9828116, 1036.9786696, 1039.9806591, 1042.9830384, 1045.9783868, 1048.9784267, 1051.9817188, 1054.9790514, 1057.9811651, 1060.9803885, 1063.9824963, 1066.9786693, 1069.9804061, 1072.9815887, 1075.979598, 1078.9787345, 1081.9841244, 1084.9798877, 1087.9783554] 	           linespace = [ 40.  41.  42.  43.  44.  45.  46.  47.  48.  49.  50.  51.  52.  53.   54.  55.  56.  57.  58.  59.  60.  61.  62.  63.  64.  65.  66.  67.   68.  69.  70.  71.  72.  73.  74.  75.  76.  77.  78.  79.  80.  81.   82.  83.  84.  85.  86.  87.  88.  89.  90.  91.  92.  93.  94.  95.   96.  97.  98.  99. 100.] 	  linespacethreshold = 0.5 	                  ld = None 	                  lt = None 	                  ii = 0 	                   t = 70.0 artisanlib\main.py::digitize:21845 	                   v = 70.0 	                  ls = [ 40.  41.  42.  43.  44.  45.  46.  47.  48.  49.  50.  51.  52.  53.   54.  55.  56.  57.  58.  59.  60.  61.  62.  63.  64.  65.  66.  67.   68.  69.  70.  71.  72.  73.  74.  75.  76.  77.  78.  79.  80.  81.   82.  83.  84.  85.  86.  87.  88.  89.  90.  91.  92.  93.  94.  95.   96.  97.  98.  99. 100.] 	              coarse = 0 	                   i = 40 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to events/quantifiers
2. Click on the "Apply" button
3. 3. See the error
[... cut ...]
[Artisan OVELHA Masteratto 27.09.22.zip](https://github.com/artisan-roaster-scope/artisan/files/10457772/Artisan.OVELHA.Masteratto.27.09.22.zip)

Please attach any relevant Artisan *.alog profiles.
don't know if it's relevant, but it was open (:
[AZ_CNF008_2021_23-01-18_1528.zip](https://github.com/artisan-roaster-scope/artisan/files/10457779/AZ_CNF008_2021_23-01-18_1528.zip)

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2023-01-19:
A typo that was introduced in Artisan v1.3.0 in 2018. Fixed by Commit 1e81127320fb5904810e6254247a098c58723fe1

closed 2023-01-19 by MAKOMO

MAKOMO, 2023-05-12:
@renatocipriani : we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave