artisan-roaster-scope/artisan #1080: Alarms with 0 in the time field should trigger

gitHub-pbMay2021, 2023-01-22:
## Expected Behavior
Alarms with 0 in the time field should trigger, tested with CHARGE and TP

## Actual Behavior
Alarms with 0 in the time field do not trigger, but alarms with non-zero in the time field do.
Alarms with non-zero (e.g. 00:15) in the time field do.

## Steps to Reproduce the Problem

  1. set up simple alarms, e.g.
  2.     Nr=1, Status=on, If Alarm 0, But not 0. From CHARGE, time 0.0, Source blank, Condition above, Value 0.0, Action Pop UP, Beep checked, Description "Charge t=0"
  3.    Nr=2, Status=on, If Alarm 0, But not 0. From CHARGE, time 0.15, Source blank, Condition above, Value 0.0, Action Pop UP, Beep checked, Description "Charge t=15"
  4. then run a test-roast on a HotTop and the popup from step (3) appears but the one from step (2) does not.
  5. Same behavior with TP instead of CHARGE

## Specifications

  - Artisan Version: 2.8.2 (also tested 2.6.0, same behavior)
  - Artisan Build (number in brackets shown in the about box): f525b16
  - Platform (Mac/Windows/Linux + OS version): Apple M1 MacBook Pro, 12.6.2
  - Connected devices or roasting machine: Hottop KN-8828B-2K+ 

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

*** Save settings is greyed out
*** .alog file attached as text

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[Charge_and_TP_alarm_trigger_test.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/10475673/Charge_and_TP_alarm_trigger_test.alog.txt)

MAKOMO, 2023-01-23:
Thanks for your report, but you hit a "feature". Actually more a compromise to avoid having another widget that decides if the time criteria is active or not. We decided, and documented in the [post introducing Alarms](https://artisan-roasterscope.blogspot.com/2013/03/alarms.html), that the time criteria is only used if not equal to 00:00. Thus you need to set the time criteria to at least 00:01, delaying the alarm by a second (or at least one sample).

closed 2023-01-23 by MAKOMO

gitHub-pbMay2021, 2023-01-23:
Thank you, I had read "post introducing Alarms" and just read it again. From reading that, having time = 00:00 means that the time condition is ignored -- makes perfect sense.  
But if that were the case, the event TP with time = 00:00 should trigger immediately when TP happens -- that is what I tested, and it doesn't do that.

Maybe the case I described is covered under "Remarks" in that documentation:
    not all events that you can select to restrict the active period of a roast via the 
    From attribute, like DryEnd, are set automatically by Artisan during a roast.....
and TP and CHARGE are both automatically set by Artisan <-- is this why I need time # 00:00?

MAKOMO, 2023-01-23:
Exactly! "From" is not a trigger. There is only a time and a temperature trigger. Both are connected via a disjunction ("or") and each of the two can be deactivated (time by using 00:00 and temp by selecting the empty source). The "From" part is just an additional pre-condition.