artisan-roaster-scope/artisan #1084: Phidget IO 8/8/8 Failed to set(1, 0)

BitterAndReal, 2023-01-30:
I often experience that the Phidget IO 8/8/8 is not connecting correctly after ON or START.
Sometimes only the digital output "set(1, 0)" is not working and sometimes the temperature readings from the thermocouple phidget are also not working.
Usually the issue is resolved after pressing OFF wait a few seconds and press ON/ START again and sometimes a couple of tries and USB replug is needed. If I don't wait for a few seconds after OFF I almost never get it working.

### Setup:
Diedrich IR-12
Phidget IO 8/8/8 connected to macbook pro with USB
Phidget Temperature Sensor 4-input connected to Phidget IO 8/8/8 with USB
Phidget Analog 4-output connected to Phidget IO 8/8/8 with USB
2x MODBUS variable frequency drives connected with FTDI USB to RS485 Serial Converter

I use Artisan [v2.8.2] with [macOS 12.2.1] but had the same issue with [v1.6] and all earlier Artisan and OSX versions.

Has anyone experienced similar issues with these phidgets?
Is it possible to acces a log with more details what is happening during initialisation of the phidgets?

[image]

[image]

[artisan-settings-20230110.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/10536212/artisan-settings-20230110.aset.zip)

MAKOMO, 2023-01-30:
Sounds like a communication problem between your Phidgets and its driver, not like an Artisan issue. Are you running the latest Phidgets driver? If not install it and use open your Phidget Control Panel app to check if all your modules are at the latest firmware version. The ones that need to be updated are listed in red. Click them to update their firmware. Then test if this improves stability. Use USB of minimal length to minimize communication issues caused by EMI from the motors of your machine.

BitterAndReal, 2023-02-01:
Thanks for the quick reply. I updated the Phidget Control Panel and driver and slightly changed the routing of the USB cable to stay away from the VFD's and motors.
I hope this solves the issue...

MAKOMO, 2023-02-05:
Please report if your measures helped such that I can close the issue here.

BitterAndReal, 2023-02-07:
We had two roasting days and it works perfect. Thanks!

MAKOMO, 2023-02-07:
Thanks for this confirmation!