artisan-roaster-scope/artisan #1085: Fuji PXF4 setup issues

Dgocof, 2023-01-31:
My Fuji PXF4 Setup.      SEE PDF Below for screen shots.

[Fuji.Setup.pdf](https://github.com/artisan-roaster-scope/artisan/files/10541130/Fuji.Setup.pdf)

On Mac with Artisan 2.8.2
We are roasting on a Probat L25 with a maxtrol modulating gas valve. The signal is setup for current as the maxtrol is set for 4-20 Ma
Phidget TMP 1101 4 Thermocouple. 2nd BT Thermocouple and 3rd Thermocouple is environment Temp
I have 2 bean probes.     1 is 1mm and 2nd is 3 mm.
3 mm is hooked to BT from Fuji. Main Thermocouple.

My Problem: See PDF below.
This is my setup but I am not getting any control of the gas valve or flame either manual with a slider or with follow background.

I am mostly following the setup: https://www.bigloose.com/v11/index.php/home/archive/artisan-roasting-with-the-pid-using-bt-as-input-fuji-pxf

I don’t seem to be communicating with the Modbus registers to operate the gas valve?
With PID on, in Control, it doesn’t seem to matter.
I think the computer is communicating with the Fuji as I get BT. When I use the  scan in port configuration I see no numbers coming up.

More Questions: See PDF Below
I’m curious what the difference is between Modbus and Modbus 34  in extra devices. Are these the 4 modbus channels?

I am also wondering if I choose PID (device Assigment) and Fuji. Can I control the gas valve with BT or only ET?

I’m also a bit confused with choosing meter (Modbus) in device assignment is that  associated with serial and input 1-8? Top half of diagram below (Port Configuration)?

If I choose PID in device assignment that is referencing PID,in (port configuration) slave,PID. Button half of diagram below?

My setup is in this pdf. I wasn't able to post my screen shots here in this screen.

[Fuji.Setup.pdf](https://github.com/artisan-roaster-scope/artisan/files/10541130/Fuji.Setup.pdf)

bigl00z3, 2023-01-31:
Dgocof,

this is most likely not an artisan issue but more a configuration issue. You should post this in the [Disscussion](https://github.com/artisan-roaster-scope/artisan/discussions) page.

> I think the computer is communicating with the Fuji as I get BT. When I use the scan in port configuration I see no numbers coming up.

If you see the ports being listed during a scan, you are communicating with the Fuji.

> I’m curious what the difference is between Modbus and Modbus 34 in extra devices. Are these the 4 modbus channels?

Modbus is for inputs 1 and 2 in the ports window while Modbus 34 is for inputs 3 and 4. Each input is talking to one of the register on the Fuji (either to read or send a value). For example 2000 is the the PV value measure by your PID.  240 is the SV value (from PID pallet 1).
When you call for Modbus in the `Device Assignment/Meter,` it will look for Modbus inputs 1 and 2.

> My setup is in this pdf. I wasn't able to post my screen shots here in this screen.

it looks like your SV BT is not the wrong register. Your Modbus 3 is set to register is set to 2041 which is Duty%. SV BT is register 240. In your screenshot the value is -5 which is the minimum value sent. It can send from -5% to 105%
[image]

For the list of registers, you can check the [Fuji communication document](https://felib.fujielectric.co.jp/download/ContentDownload.htm?dataid=16281277&site=global&lang=en)

You can also check my 1st post when I started using the PID [Artisan – Roasting with the PID – Fuji PXF (and test of MaxWell MTA)](https://www.bigloose.com/v11/index.php/home/archive/artisan-roasting-with-the-pid-fuji-pxf-and-test-of-maxwell-mta) they are some aditional information to the post you have been following.

> I am also wondering if I choose PID (device Assigment) and Fuji. Can I control the gas valve with BT or only ET?

Initially I used to control via ET and then moved to controling from BT. Both are possible but as the processing is done by the Fuji itself, you need to connect (physicaly) the right probe to it (either ET or BT). If you want to control with the other probe you have two options:

- Add a second Fuji PXF (it will need to be slave 2), the 1st one connected to ET and the second one connected to BT.
- When you want to change probe without adding a second Fuji nor rewirreing, you could use the Artisan internal PID (slower). For this change, slave number from one to zero.

Nico

MAKOMO, 2023-04-18:
Stall and most likely not an Artisan issue. Closed.