artisan-roaster-scope/artisan #1087: Santoker R500 Bluetooth, not connecting on Artisan.

acx4, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

MAKOMO, 2023-02-01:
Not much details in your report;) The Santoker R500 connection is only supported via WLAN.

closed 2023-02-01 by MAKOMO

acx4, 2023-02-01:
HI Makomo, I made another report, the is R200 Bluetooth which is working on the Santoker app, it's my sample roaster and I would like to use Artisan for both the sample and the production roaster.  The Santoker App works on Bluetooth in Windows 10 and Linux Ubuntu 22.04 LTS, but Artisan does not.  I also have a smaller Q5 2.0  which that does work via WiFi on both the Santoker App and Artisan.

MAKOMO, 2023-02-01:
Artisan is not support Bluetooth. Sorry. Thus at any time we accept PRs. Just send in your code...