artisan-roaster-scope/artisan #1088: Santoker R500 Bluetooth not working on Artisan 2.8.2

acx4, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
Running Linux, Ubuntu I cannot connect artisan using Bluetooth, LInux sees the Santoker machine, and all permissions are correct.  The Santoker software sees it fine on the same computer.  

**Describe the solution you'd like**
I would like the Artisan software to recognize the Santoker R500 Bluetooth to use that software across all machines.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Using Linux, also tried in Windows 10 but only Santoker software works on both, not Artisan software.

acx4, 2023-02-01:
HI Makomo, I made another report, the is R200 Bluetooth which is working on the Santoker app, it's my sample roaster and I would like to use Artisan for both the sample and the production roaster. The Santoker App works on Bluetooth in Windows 10 and Linux Ubuntu 22.04 LTS, but Artisan does not. I also have a smaller Q5 2.0 which that does work via WiFi on both the Santoker App and Artisan.

acx4, 2023-02-01:
Just to Clarify there are two R200, the R200 and the R200 Bluetooth, Cropster has just fixed this issue, so I am hoping it is possible to fix with Artisan.  I can test for you if possible.  Thank You.

MAKOMO, 2023-02-02:
Artisan is simply not implementing the Bluetooth communication. I also don't have a device to test thus I will not implement this. If you can submit an implementation of such a Bluetooth connection, please submit its code as PR and we will integrate it if we find the resources to do so. Note that Santoker never supported the Artisan project in any way and nobody in the team is not willing to do commercial software development for free.

MAKOMO, 2023-04-23:
I just added a variants of the "Santoker Q Series" "Santoker R Series" which communicate the very same protocol via Serial/Bluetooth (not BLE!) instead of Network/Wifi. However, I am not sure this works for you as I don't have any technical information on this connection. Especially, I do not know the serial parameters (and if the transport is indeed serial communication via Bluetooth and not BluetoothLowEnergy BLE).  Thus you need to specify the correct comm port (connected via BT) and the baudrate, byte size, parity and stopbits under menu Config >> Ports, ET/BT tab.

The new continuous build with this addition is available here:

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Please report back if this works for you and if it works please tell me the correct serial parameters.

WuTr3YZd, 2023-12-19:
Unfortunately, the aforementioned devices all work in BLE mode.

MAKOMO, 2024-10-17:
[Santoker R Series support](https://artisan-scope.org/machines/santoker/) was just improved and Bluetooth communication was added. Could you please check if the setup `R Series Bluetooth` (menu `Config >> Machines >> Santoker`) of the current [Continuous Build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) works with your machine (assuming it is not a R Master Series and it features Bluetooth communication).

MAKOMO, 2024-11-06:
Stall

closed 2024-11-06 by MAKOMO