artisan-roaster-scope/artisan #1092: Each environment variable has trailing \n\n when Artisan invokes scripts

clarsen, 2023-02-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. In terminal, 
PYRO_NS_HOST=192.168.6.85 open /Applications/Artisan.app
2. From within Artisan, Invoke shell script, e..g through specifying a script to execute via Config>Events>Start
3. Use $PYRO_NS_HOST in a script.

**Expected behavior**
Environment variables should be verbatim.

**Screenshots**
example dump of env within a shell script:
```
echo $PYRO_NS_HOST | od -c
```

produces
```
0000000    1   9   2   .   1   6   8   .   6   .   8   5  \n  \n
0000016
```

would not expect the two trailing \n.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.2
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: f525b16
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 12.6.2
 - Connected devices or roasting machine [e.g. Probatone 5]: SR700 via SR700_Artisan_Client -> SR700-Artisan-PDServer

**Additional context**
Doesn't happen with 2.4.6.  I couldn't get binaries of 2.6.0 to test with.

I believe this happens for all environment variables since env from within a shell script invoked by Artisan outputs each variable with newlines separating them, e.g.:
```
rvm_bin_path=/Users/clarsen/.rvm/bin

TERM_PROGRAM=iTerm.app

SSL_CERT_FILE=/Users/clarsen/Applications/Artisan-2.8.2.app/Contents/Resources/openssl.ca/cert.pem

QT_MAC_WANTS_LAYER=1

TERM=xterm-256color

SHELL=/bin/zsh
...
```

Would expect this instead:
```
rvm_bin_path=/Users/clarsen/.rvm/bin
TERM_PROGRAM=iTerm.app
SSL_CERT_FILE=/Users/clarsen/Applications/Artisan-2.8.2.app/Contents/Resources/openssl.ca/cert.pem
QT_MAC_WANTS_LAYER=1
TERM=xterm-256color
SHELL=/bin/zsh
...
```

I believe the artisanlib.calc_env() function is not stripping the newlines as it is reading lines from Popen.

**Workaround

Fortunately, now that it is clear what's going on, just set the variables from within the script rather than pass them in via environment.

MAKOMO, 2023-02-05:
Thanks for this detailed report. This should be fixed now by Commit 58d9b28de28b18fd4d4e0436a87f3aaf8952a752.

closed 2023-02-05 by MAKOMO

MAKOMO, 2023-02-08:
Could you verify that this issue is fixed by the current [Continuous Release](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous)?

clarsen, 2023-04-11:
> Could you verify that this issue is fixed by the current [Continuous Release](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous)?

apologies for the late response... checking shortly

MAKOMO, 2023-05-12:
@clarsen: we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave

clarsen, 2023-05-21:
I had checked out the earlier build https://github.com/artisan-roaster-scope/artisan/commit/97ec3ef9c95a0f2f986409c3961e84f5eab20131 and there seemed to be a different issue with passing arguments when using Call Program in the slider event.  however that doesn't show with this https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta version.

I didn't run a roast, but ran with test settings and alarms and a test shell script that logged environment and arguments to a log file.

Both the environment variables passed into the various Call Program invocations (buttons, slider events, device ET/BT external program) and the arguments to the program themselves seem to be working properly now.

Thanks!