artisan-roaster-scope/artisan #1094: Time Discrepency between artisan and Artisan Plus

patioroaster, 2023-02-06:
**Describe the bug**
30 minutes is being added to my roast times when viewed in Artisan Plus.  When I go to update my roast in Artisan Plus the Development time wqill read at 31:50 (as an example)  And total roast time will be in the 40 minute range.

**To Reproduce**
Steps to reproduce the behavior:
Unsure how to reproduce

**Expected behavior**
The time from my roast in Artisan should be reflected in Artisan plus

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version 2.8.0 *HOWEVER* Has been present much longer and on previous builds
 - Connected devices or roasting machine: Phidgets (Vint Hub (current firmware and previous) with 1100 or 1101)  Aillio Bullet (Current firmware and previous)
 -  Systems - Windows 11 (Multiple Machines), Linux (Pop OS and Raspian
 - 
 

**Additional context**
Seems to have been present for my whole time roasting with Artisan.
I am in Adelaide with time zone set as +9.5 (Not sure if relevant)
[Guatemala Antigua_23-02-06_1012.zip](https://github.com/artisan-roaster-scope/artisan/files/10632542/Guatemala.Antigua_23-02-06_1012.zip)

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog
[artisan-settings-patio-phid-Toper.zip](https://github.com/artisan-roaster-scope/artisan/files/10637068/artisan-settings-patio-phid-Toper.zip)

 
[image]

MAKOMO, 2023-02-12:
This is not an issue with the Artisan software, and not even a problem with the data itself. It turned out that just under some circumstances the rendering of those times on the artisan.plus platform is done wrong. We have fixed this and will deploy the fix next week. Thanks for reporting and sorry for the confusion.

patioroaster, 2023-02-13:
Thanks very much for the quick resolution.  I’m glad it was a simple fix!

MAKOMO, 2023-04-03:
resolved

closed 2023-04-03 by MAKOMO