artisan-roaster-scope/artisan #1100: MODBUS Scan option broken in v2.8.2?

bw00st3r, 2023-02-15:
## Expected Behavior
MODBUS Scan should display values stored in all registers. It worked with v2.6, and I could see all the register values, and that was how I figure the correct register to use.

## Actual Behavior
In v2.8.2 the information box remains empty.

## Steps to Reproduce the Problem

  1. Artisan 2.8.2 -> Config - Device
  2. ET/BT -> Meter: MODBUS -> OK
  3. Ports Configuration -> MODBUS -> Scan -> Start
  4. No data is written into the window
  
## Specifications

  - Artisan Version: 2.8.2
  - Artisan Build (number in brackets shown in the about box): f525b16
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: MODBUS via USB

MAKOMO, 2023-02-15:
Duplication. See Issue #1056.

bw00st3r, 2023-02-15:
agreed. closing.

closed 2023-02-15 by bw00st3r