artisan-roaster-scope/artisan #1101: 2.8.2 PID on ET follow BT background instead ET

ipa64, 2023-02-16:
Updated to 2.8.2 from 2.8.0
Go for a new batch with my hot bed roaster, BT on channel 1 ET on channel 2, Control PID channel 2.
Same background profile and all settings as before, no change only updated from 2.8.0 to 2.8.2

START
Press CHARGE
ET temperature follow BT background line !
STOP
RESET, reboot my MacBook
Try again, same behavior (3 attemps)

Downgrade to 2.8.0
Start again, all works as expected, as all previous roast

MAKOMO, 2023-02-16:
You did not tell us which setup you are using. I guess you are using a Arduino/TC4 setup. For this an issue in the source channel configuration of its PID configuration dialog was introduced in v2.8.2 and fixed by now. The fix is already integrate in the current [Continuous builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).

ipa64, 2023-02-16:
Yes you are right, I use an Arduino/TC4 setup.
Thank you I will try with this build on the next batch

closed 2023-02-16 by MAKOMO

ipa64, 2023-02-20:
This morning I tried a roast with version 2.8.3 , I got same wrong behavior
[https://i84.servimg.com/u/f84/20/23/21/77/captur29.png](url)