artisan-roaster-scope/artisan #1110: 2.8.2 & 2.8.3 PID on ET follow BT background instead ET

ipa64, 2023-02-21:
Original report was #1101
Updated to 2.8.3 from 2.8.0 on MacBook Pro
Hot bed roaster, Arduino/TC4 
BT on channel 1 ET on channel 2, Control PID channel 2.
Same background profile and all settings as before, no change only updated from 2.8.0 to 2.8.3 (or 2.8.2)

START
Press CHARGE
ET temperature follow BT background line !
STOP
RESET, reboot my MacBook
Try again, same behavior (3 attemps)

Downgrade to 2.8.0
Start again, all works as expected, as all previous roasts

below the print screen of the aborted roast tested on 2.8.3
https://i84.servimg.com/u/f84/20/23/21/77/captur29.png

MAKOMO, 2023-02-25:
Could you please attach your Artisan settings file (save to artisan-settings.aset via menu Help >> Save Settings; rename into artisan-settings.txt and drag-and-drop it to this page on Github into a comment box). Thanks!

MAKOMO, 2023-02-25:
I might have found the issue. Could you test the current Continuous Build and report back?

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

ipa64, 2023-02-27:
Ok I will test it tomorrow, thank you for your support

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/10836219/artisan-settings.txt)

ipa64, 2023-03-08:
Solved (pre-release 2.8.3.dmg)
Thank you

closed 2023-03-08 by ipa64

MAKOMO, 2023-05-12:
@ipa64 : we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave

ipa64, 2023-05-14:
Hi,
I have just roasted 2 batches with this beta.
Setup: fluidbed roaster, Arduino/TC4, PID on ET, Artisan running on MacBook Pro ).
All has worked as expected, perfect!
Thank you

Have a nice day
Patrick

MAKOMO, 2023-05-14:
Excellent! Thanks for testing and reporting.