artisan-roaster-scope/artisan #1124: Loss of Power Control in Background Mode

stephenpate, 2023-03-12:
**Describe the bug**
I'm new to the Artisan software, but a buddy of mine had the same issue today. When following a roast profile from loading a background roast, the software takes away power control if you come in TP a little hot. I made me stay at 0 power until the roast hit the curve, but by then the roast was stalling. 

**To Reproduce**
Steps to reproduce the behavior:
Come in to TP high and try to increase power prior to curve data. 

There is no option for PID on or off, so I don't think it in auto mode because I can control the rest of the roast. Thank you!

Stephen.Pate@hotmail.com

bigl00z3, 2023-03-12:
This is not an artisan issue or a bug. It is the way the PID is reacting based on your configuration.

You need to first have the good P, I and D values for your roaster and then you also need to configure everything around in Artisan to work well. Hoping to use the PID function without a proper configuration will almost for sure not work properly. If you are too hot and the background curve is lower, Artisan PID will remove power to reach the background.

You can read about PID and Artisan on:

-  [Artisan Blog](https://artisan-roasterscope.blogspot.com/2016/11/pid-control.html)
- [in the discussion list ](https://github.com/artisan-roaster-scope/artisan/discussions?discussions_q=PID)by searching for PID

Nico

stephenpate, 2023-03-12:
Thank you! Figured it was user error. I'll do some more reading!

bigl00z3, 2023-03-13:
@stephenpate  Until you are done reading, you can close this issue. 
And if you have more to add or questions related to this topic use the [Discussion thread](https://github.com/artisan-roaster-scope/artisan/discussions/1125) instead.

Nico

closed 2023-04-03 by MAKOMO