artisan-roaster-scope/artisan #1126: slider selection highlight doesn't work when sliders are all the way up

tkdrumr, 2023-03-14:
**Describe the bug**
The slider bar is highlighted when selected - only in the bar above the slider. The problem is that when the slider is all the way up it does not show a change in the bar that indicates it is the selected bar. This is easiest to see when using the tab key to go between bars and the arrow keys to move the slider up and down. When the slider is in the center you can see that the bars above the sliders change color or tone but below the slider remains unchanged.

**To Reproduce**

Steps to reproduce the behavior:
1. Open Artisan (both sliders bars are not selected - sliders at lowest level)
2. Press the tab key (the left slider bar will be highlighted)
3. Press the tab key (the right slider bar will be highlighted)
4. Press the up arrow to move the slider to the middle of the bar
5. Press the tab repeatedly and note the change in color/tone on the bar that is at the midpoint. Only the top continues changing while the bottom of the bar remains a constant color.
6. Press the tab and move the other slider all the way to the top
7. Press the tab again and note that when the bar with the slider at max is selected there is no change in color/tone
8. Press tab and move the other slider all the way up so they are both at max ... pressing tab has no effect on the screen. You cannot tell which bar is selected.

**Expected behavior**
The selected bar should show some sort of indication it has been selected - perhaps bolding the bar title, placing a highlight around the entire bar, or a box around the bar value. The bar values have different colors. Maybe that isn't necessary and you can change the color only when that bar is selected.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [V2.8.2]:
 - Artisan Build (number in brackets shown in the about box) [f525b16]:
 - Platform (Mac/Windows/Linux + OS version) [Windows 10]:
 - Connected devices or roasting machine [Phidget tmp1101_1/VINT HUB0001_0]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-settings - Copy.txt](https://github.com/artisan-roaster-scope/artisan/files/10973037/artisan-settings.-.Copy.txt)

MAKOMO, 2023-04-12:
Thanks for your clear report. This has been fixed. The fix will be part of the next release.

closed 2023-04-12 by MAKOMO

MAKOMO, 2023-05-12:
@tkdrumr: we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave

MAKOMO, 2023-05-12:
Thanks a lot! Awaiting your report...