artisan-roaster-scope/artisan #1136: Sliders command is not executed automatically in Artisan PID mode

peterweiser, 2023-03-29:
I wanted to use a slider to control the burner power. Therefore, one slider was configured to control a pwm device output (vint phidget rel_1100), which was connected to a proportional gas valve. I used the PWM command "out(0,{})" for this slider. This system itself works quite well - if I use the Phidget control panel as well as if I use the configured slider in Artisan. The burner behaves like expected - the power of the burner is controlled in relation to the slider position.

Describe the bug**
The behaviour is that if I move the slider and release it at any position, the burner power is controlled as expected in relation to the slider position unlike as if Artisan software-PID controls the slider automatically, I can see the slider moving, and the value changing. The problem is, that if the artisan PID works, the PWM signal does not change even if the slider moves correctly according to the background profile. So there is no real action at the burner visible, if the slider is controlled automatically with the artisan PID. It´s like if the configured PWM command would not work, but as mentioned it does if the slider is controlled manually.

**Expected behavior**
If Artisan PID controls the slider, the slider should move and the values should change as they already do, but also the PWM signal should change too. 

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: same behavior in 2.4.6 (which I am using)  or 2.8.0 or 2.8.2
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: f1b8e57
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10 Pro
 - Connected devices or roasting machine [e.g. Probatone 5]: Dätgen DR12, 2 Delta DTB4848 PIDs (which I want to substitute with the artisan software PID) connected via modbus , Phidget VINT HUB with 2 Thermocouples and a solid state relay module

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[artisan-einstellungen_phidget-control_3f_2023.txt](https://github.com/artisan-roaster-scope/artisan/files/11103061/artisan-einstellungen_phidget-control_3f_2023.txt)

MAKOMO, 2023-04-17:
Artisan supports a number of different PIDs. One is its internal PID, however, it can also drive the external Arduino/TC4 PID, the FUJI/DTA PIDs, or other external PIDs connected via the MODBUS or S7 protocols. The PID dialog looks in all cases very similar (next version will indicate which setup is active).

Your Artisan settings are configured to address an external PID via MODBUS. That way Artisan is only sending p-i-d configurations, PID ON/OFF messages and SV messages to that external device as configured and assumes that its PID loop drives the actor accordingly.

To disable that external MODBUS PID configuration you need to put a zero as its MODBUS slave ID. Then the internal Artisan Software PID is activated (assuming you did not configure for a S7, Fuji/DTA or Arduino/TC4 PID; which you did not according to your your settings).

Still your configuration is a little wired as your burners generates PWM signals only between 76-86% due to its slider offset/factor configuration. You also ticked "Inverted Control" in the PID dialog which does not make much sense here.

Otherwise I tested the PID in combination with a PWM Phidget and it operates as intended. Thus I close this issue.

MAKOMO, 2023-04-17:
Could you let us know why you are you using 2.4.6 instead of the latest version of Artisan?

closed 2023-04-17 by MAKOMO

peterweiser, 2023-04-18:
Hi Marko

thank you very much for taking so much time to analyze my problem.

First of all, I want to point out that I m very happy with the artisan software and all that cool functions. I m successfully using it at the roastery since 3 years.

I read that, but I didn´t really reallize that I have to set the Modbus slave id to 0 to enable the artisan software PID.
I have configured it that way because I wanted to use that Delta PID only to get the values of the temperature sensors in combination with the Artisan Software-PID with PWM Control. But in future, instead using the temperature sensors which are connected to the Delta PID, I ll use thermocouples connected to a phidget, which I ve already bought for future use.

I found out, that the slider offset works best with that gas-valve if I use a PWM range between 76% and 86%.
Inverted control was only enabled for testing.

I am using version 2.4, because the behaviour of the newer versions changed. In version 2.4, I can use the configured alarms, to control the drum temperature between the batches to set it to for example 250 degrees for 1 minute to heat the roaster up after one batch was finished and to set it to 220 degrees back for the next roast. Sinse version 2.6, that did not work anymore and unfortunately, I didn´t find out the reason. Artisan uses the load temperature of the background profile also between the batches and doesn´t follow the alarm temperature rules anymore.
I tried every single new version until 2.8.2, but this behaviour did not change. 

In version 2.8.2, I am also experiencing the modbus communication error "Durchstanzen" described a few days ago.

MAKOMO, 2023-04-18:
> I am using version 2.4, because the behaviour of the newer versions changed. In version 2.4, I can use the configured alarms, to control the drum temperature between the batches to set it to for [...]

Why didn't you report this before?

>Artisan uses the load temperature of the background profile also between the batches and doesn´t follow the alarm temperature rules anymore. I tried every single new version until 2.8.2, but this behaviour did not change.

Indeed. However, your alarm rules work as before, their effect is just overwritten by this new behavior. As a simple workaround, just put the PID from background follow into manual mode. This can be done either via alarm rules which trigger (hidden) custom buttons that modify the PID follow mode or by adding corresponding actions directly to the ON and START buttons. 

a) using two custom event buttons

[image]
[image]

[artisan-settings-bbp-custombuttons.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/11261774/artisan-settings-bbp-custombuttons.aset.txt)

b) using main event button actions

[image]
[image]

[artisan-settings-bbp-eventbuttons.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/11261775/artisan-settings-bbp-eventbuttons.aset.txt)

> 
> In version 2.8.2, I am also experiencing the modbus communication error "Durchstanzen" described a few days ago.

Maybe you can help me with a proper DEBUG log using Artisan v2.8.2 in debug log mode to investigate those. I still do not received enough details to start even the investigation. Without being able to understand or best being able to reproduce this there is no chance to get it ever fixed.

peterweiser, 2023-04-20:
Wow - thank you for that solution Marko! I ll implement and test it as soon as possible.
I didn´t report the changed behaviour because I wanted to double-check first, if it´s behaving that way because I configured something wrong or not. 
[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/11286672/artisan.log)

Of course I try to help debugging this modbus issue, I ve attached a log with debug-mode enabled.  I hope, that It helps somehow.

peterweiser, 2023-04-20:
I ve just tried the settings you suggested and it works great - I m now successfully using version 2.8.0.
Please tell me, if you need further files to debug the modbus issue.

peterweiser, 2023-04-20:
Marko - I ve just tried the configuration with slave id set to 0 and now the pwm phidget actor works perfectly - thanks again for your help!

MAKOMO, 2023-04-21:
Peter,

could you test if the current continuous builds communicate to your machine via MODBUS successfully?

https://github.com/artisan-roaster-scope/artisan/releases/continuous

peterweiser, 2023-04-22:
Hi Marko - just tried tha latest 2.8.3 continous build - now the modbus communication works again

MAKOMO, 2023-04-22:
Excellent! Thanks for testing and reporting.

peterweiser, 2023-04-22:
Hi Marko - I ve experienced another problem while using version 2.8.3 daily build. 
The modbus communication itself seems to work good for the Delta PID, but there is another problem with the drum and fan motor controls (which are also connected to MODBUS. 

I ve attachecd the debug log again, which contains entries from working version 2.8.0. Today morning (22.4. around 9:54), I upgraded to version 2.8.3 Then - later in the log, you find entries from version 2.8.0, to which I downgraded again.

MAKOMO, 2023-04-22:
Nothing attached

peterweiser, 2023-04-22:
[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/11300803/artisan.log)

MAKOMO, 2023-04-22:
Thanks for this additional clear report. I found a regression in the handling of the slider values caused by our major code rewrite. I fixed that and will commit the fix in a few minutes. New builds should be available in about an hour.

peterweiser, 2023-04-25:
thank you Marko - let me know, if I can test something again for you.

peterweiser, 2023-04-25:
Great Marko - Just tested the latest build and everything works just fine. :-)

I ve another question regarding to PID values. As I mentioned, I ve rebuilt the control of my roaster with some Phidgets, which should replace the Modbus Delta PIDs. Generally, the control of the gas valve works good now, but I ve troubles in finding the right PID values. I know, thats a bit off topic, but could you recommend me some start values for a 12kg drum roaster with atmospheric burner and a single walled drum? I thought, that similar maschines behave similar in that case. Actually, it´s really difficult to get a good result of the roasting with the new phidget control, I ve switched back to the Delta PID solution again for now. But I want to have the software PID working with my maschine. I ve tried the PID values which the delta PID found out with the auto tuning funktion, but these values do not really work for the artisan software PID. Is it possible to develop such a auto tuning funktion in Artisan too?

MAKOMO, 2023-04-25:
Peter, PID tuning is difficult and very much depends on the thermic behavior of your machine (machine size, heat capacity, burner capacity and batch size). Autotuning is difficult to apply here as it it remains unclear in which state the PID should be tuned. This very much depends on your usage of the PID. Autotuning during roasting is difficult and would waste quite some beans. To my opinion the PID is most useful for pre-heating and for the between batch protocol or any time you need to leave your machine unattended. To replicate a roast profile, replaying the recorded burner events is usually much more successful. Sorry.

MAKOMO, 2023-04-25:
But as an open-source project we are always happy about good contributions via PRs!

MAKOMO, 2023-05-12:
@peterweiser: we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave

peterweiser, 2023-05-13:
Hi Marko,
this Version (2.8.3 beta) works quite well and stable for me in my environment with modbus and the phidgets. 
But I could find 2 issues:

first one: 
When starting Artisan, in earlier versions, the last used background profile was loaded if the checkbox to do so, was checked. 
It doesn´t look like there is any change if the checkbox is checked or not - in either case, the background profile is not loaded automatically anymore. although lhe path to the last used profile is still filled in (like in my screenshot).
[image]

second one:
The "auto empty" funktion doesn´t work after every roast anymore, but sometimes it still does. The checkbox "auto leeren" was checked. Even though, Artisan does not recognice anymore when the beans are dropped.

MAKOMO, 2023-05-13:
Dear Peter, thanks for this fast review and your detailed report. Good to here that this works well with your Phidgets and MODBUS devices. Regarding the first issue: I see that you have the "Profilevorlage beim Laden eines Profils verbergen" ticked. With that all all elements of the background profile are hidden on loading it. Thus the last profile was loaded, but just not rendered in this case. Could you untick this flag and see if your background profiles not only get correctly loaded, but also displayed on profile load?

For your second issue: the "auto DROP / auto LEEREN" (not auto empty)  mechanism is based on a heuristic that tries to recognise a sudden drop in BT at the end of a roast by observing the last 5 readings (it only works in retrospect, thus about 3 reading after the case). That heuristic fails on some machines/signals and is difficult to get working reliable across all machines. Anyhow, that mechanism did not change w.r.t. the previous v2.8.2 release, but did change slightly w.r.t. v2.8.0 (if I remember correctly). Please attach one or two profiles as .alog file here (you need to rename it to .txt) where the mechanism failed (no problem if you manually added the correct LEEREN event already) and I can see if this can be improved for you.

Thanks Peter!

peterweiser, 2023-05-13:
Ah okay, but then I dont really understand the usage of that checkbox. I thought that the background profile is processed during the roastings, but only not shown if the checkbox is checked. 
Now, I ve unchecked that checkbox and now it works like I thought. 

The second issue - I could not really try these older versions extensively because of the modbus bug, but I am sure, that a changed behaviour was noticeable there already in version 2.8.0.
Before, I was using version 2.6 in my production environment and that function worked perfect and really reliable in every single roast I have made so far.

[2023-05-13_1222_Sat_2023391 (2)_Brasil Fazenda Lagoa_Brasil Fazenda Lagoa_9.5Kg_17_54m_AUC1113_Bag-A39.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/11469469/2023-05-13_1222_Sat_2023391.2._Brasil.Fazenda.Lagoa_Brasil.Fazenda.Lagoa_9.5Kg_17_54m_AUC1113_Bag-A39.alog.txt)

[2023-05-13_0939_Sat_2023389 (1)_Brasil Fazenda Lagoa_Brasil Fazenda Lagoa_9.5Kg_18_00m_AUC1141_Bag-A39.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/11469470/2023-05-13_0939_Sat_2023389.1._Brasil.Fazenda.Lagoa_Brasil.Fazenda.Lagoa_9.5Kg_18_00m_AUC1141_Bag-A39.alog.txt)

MAKOMO, 2023-05-14:
Thanks for those profiles. I will investigate...

> Ah okay, but then I dont really understand the usage of that checkbox. I thought that the background 
> profile is processed during the roastings, but only not shown if the checkbox is checked.

Exactly! The background profile is just hidden, but if background playback events mode is turned on the events from the background are still replayed.

> Now, I ve unchecked that checkbox and now it works like I thought.

As said. This check box just controls if directly after load the "Show" flag in the very same dialog is unticked automatically to hide the background profile curves. Still the background is loaded and fully "operational".