artisan-roaster-scope/artisan #1141: phidget IO command error message "Failed to set"

ivanroaster, 2023-04-02:
Artisan v2.82 (build f525b16)
Phidgets22 v1.14.20230331

Phidget REL1101_1 (16x Isolated Solid State Relay Phidget)

Artisan Events - Buttons
Action: IO command
Documentation: set(ch,{},<sn:p>), where ch is channel, sn serial number p port
Value: 1 on / 0 off

When executing the button:
message displayed in Artisan "Failed to set (ch,1,<sn:p>)"
message displayed in Artisan "Failed to set (ch,0,<sn:p>)"
and no action happens in the phidget

Notes:
the same phidget channel digital state can be changed correctly from the phidget control panel
the same phidget channel can be controlled through Artisan using 
Action: PMW command
Documentation: out(ch,{},<sn:p>), where ch is channel, sn serial number p port
Value: 100% on / 0% off

**Expected behavior**
set on or off the digital state of channel ch in module attached to port p of hub sn

MAKOMO, 2023-04-18:
By design each physical Phidget channel can be used only in one way (accessed via one API). So either that solid state channel is used via the PWM API or the IO API. 

I assume you first used a PWM Command which reserves the channel for PWM access. Then you tried addressed it via an IO Command which failed as the (hardware) channel was already in use (by the PWM API).

If you would restart Artisan (to free the channels) and first send an IO Command it should work, but later PWM Commands on the same channel will fail.

This is a technical restriction which is maybe avoidable in this case, but not for channels like HUB ports that can act as analog/digital input or output.

Can you confirm?

ivanroaster, 2023-04-22:
Hi Marko,
Thanks for your reply.
Today I created a new button and tested what you mentioned but unfortunately, no, I keep getting the failed message with IO commands even when I first run this command before anything else after restarting Artisan using REL1101_1 and a phidget hub. With the PWM commands I can do an out and toggle functions with no problem.
Is this functionality tested with this phidget module and current phidgets22 api?
Thank you
Ivan

MAKOMO, 2023-04-23:
Dear Ivan,

Yes, this is tested and working using a REL1100_0.  

[image]

I don't have a REL1101_1. Just found a typo in the code responsible for discover IO Phidgets and it was looking twice for REL1100, but never for REL1101. Sorry. Fixed that. Please download and install the continuous build (should be online in a few minutes) from here

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

If that works successfully for you, please confirm such that I can close this issue here.

ivanroaster, 2023-04-28:
Hi Marko, I just tested it with the continuous release link you sent me and it works perfectly. Thank you for the fix!

closed 2023-04-28 by ivanroaster

MAKOMO, 2023-05-03:
I just joined those two APIs in my latest commit. Now you should be able to use both APIs on the same attached module in parallel. Thanks for the hint!

MAKOMO, 2023-05-12:
@ivanroaster: we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave