artisan-roaster-scope/artisan #1144: Artisan cannot connect to Hottop KN-8828B-2K+

MarcelPr, 2023-04-05:
Every time I turn my (old) Hottop on, Artisan cannot connect to it and cannot read ET and BT and is unable to control it.
I found a fix. I need to set up machine first, but I choose TC4 first. Then I set up machine again, choosing KN-8828B-2K+.  By default the baud rate is  12600, and I need to keep it. After that I click "ON" and now artisans reads ET and BT, although incorrectly. After that I keep the machine and device, but change the baud rate to 9600. After that Artisan works as expected. But next time the story continues.

Any comments or advise? I could do the above procedure every time, no problem. But I thing it might be a bug to fix.

Marcel

MAKOMO, 2023-04-06:
Marcel, which version of Artisan are you using?

MarcelPr, 2023-04-06:
2.8.2

MarcelPr, 2023-04-06:
Hi Marco, for your convenience I recorded my screen. I illustrate 3
attempts to control hottop.
1) I start Artisan with corect settings and it does not work.
2) I start and configure Artisan correctly starting from "machine". Does
not work.
3) I start Artisan and configure it incorrectly. The baude rate is 11xxx.
Then I hit Control. It now controls Hottop.
    Only after that I can change the baude rate to 9600 and it finally
works.

Here is the link
https://drive.google.com/file/d/1UdMr-CmktTn_ZGia2q1kNHiRTUAAqIGr/view?usp=sharing

MAKOMO, 2023-04-12:
The correct baud rate for the Hottop KN-8828B-2K+ is 115200 (as you demonstrate around 2:09). Not sure why you switch in your video around 01:00 to 9600 which obviously does not work. That the connection fails after switching back to the correct 115200 is because the serial port object once established by pressing ON is not cleared on OFF nor renewed. The reason is that the connection to the machine is kept such that control commands can be send after OFF. This you would need to restart Artisan after switching the serial port speed to see an effect. I changed this behavior now for the next version such that changing any port setting will disconnect the serial port such that it a connection is reestablished using the new settings on ON.