artisan-roaster-scope/artisan #1145: Modbus com error

Durchstanzen, 2023-04-09:
Hi.
I updated my Artisan software from 2.4 to 2.8.2 on a windows 10 machine. Prior to that it did work with a custom Modbus roast project pretty well, but since then If connected to correct usb port it only jumps between Modbus connected and Modbus failure.

I used an other laptop with 2.4 on it and it works like a charm.

there must be a bug in 2.8.2 regarding Modbus connections…

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]:
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:
 - Connected devices or roasting machine [e.g. Probatone 5]:

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

MAKOMO, 2023-04-12:
Not enough information. We need to be able to reproduce this to allow us to investigate.

MAKOMO, 2023-04-12:
At least your artisan settings. We do not have the resources to rebuild your Arduino custom client, but maybe you can attach us log files. If you hold the ALT key while clicking on the plus icon in the upper left corner an email with the Artisan log files attached should open. Best to record the communication issues while in debug logging mode. Toggle the logging mode by holding the ALT+CONTROL keys while clicking on the plus icon.  The serial log  (menu Help/Serial) might be of help as well. Thanks!

Durchstanzen, 2023-04-12:
Will try to get that for you.

joeky888, 2023-04-17:
Same here, reverting to 2.4 solved the issue

MAKOMO, 2023-04-17:
Without information that makes this reproducible for us we will never be able to fix this.

MAKOMO, 2023-04-17:
This is not very productive. We need to know what you are sending and how you configured Artisan. Please attach your artisan-settings.aset file here (by rename to artisan-settings.txt and drag-and-drop here on a comment window). You can save this via menu Help >> Save Settings.

What MODBUS server are you running? Is that server available in source code?

Please attach also the Artisan artisan.log log file here (again renamed to *.txt and added via drag-and-drop to a Window here on Github). Record first a bit of that communication with debug logging turned on (click plus icon while holding the ALT and CONTROL keys to toggle logging mode). You get the log file attached to an email message by clicking the plus icon while holding just the ALT key.

Thanks!

joeky888, 2023-04-17:
> What MODBUS server are you running? Is that server available in source code?

Kapok 501, MODBUS 34. If you need other info just name it, I will keep tracking this issue, thank you.

[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/11251572/artisan.log)
[artisanViewer.log](https://github.com/artisan-roaster-scope/artisan/files/11251573/artisanViewer.log)
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/11251574/artisan-settings.txt)

Durchstanzen, 2023-04-17:
Where do I find these files? Turned on logging but don’t see an artisan.log…

joeky888, 2023-04-17:
And here is the debug log

```
04-17 23:17:30 MainThread artisanlib.main      INFO   : memory used 117.8M, 7.7G (65%) available
04-17 23:17:30 MainThread artisanlib.main      INFO   : ON MONITOR (sampling @3.0s)
04-17 23:17:32 Dummy-3    artisanlib.modbusport INFO   : invalidResult(1) => pymodbus error: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
04-17 23:17:33 MainThread artisanlib.main      INFO   : OFF MONITOR
04-17 23:17:33 Dummy-3    artisanlib.modbusport INFO   : invalidResult(1) => pymodbus error: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
04-17 23:17:39 MainThread artisanlib.util      INFO   : debug logging ON
04-17 23:17:39 MainThread artisanlib.main      DEBUG  : message: debug logging ON
04-17 23:17:43 MainThread artisanlib.main      DEBUG  : restoreExtraDeviceSettingsBackup()
04-17 23:17:43 MainThread artisanlib.main      INFO   : memory used 118.2M, 7.8G (65%) available
04-17 23:17:43 MainThread artisanlib.main      DEBUG  : message: Scope has been reset
04-17 23:17:43 Dummy-4    artisanlib.modbusport DEBUG  : readActiveRegisters()
04-17 23:17:43 Dummy-4    artisanlib.modbusport DEBUG  : connect(): connecting
04-17 23:17:43 MainThread artisanlib.main      INFO   : ON MONITOR (sampling @3.0s)
04-17 23:17:43 MainThread artisanlib.main      DEBUG  : message: Scope monitoring...
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : updateActiveRegisters()
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : active registers: {3: {2: [18176], 1: [18176], 3: [18176]}}
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : clearReadingsCache()
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : clearReadingsCache()
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : readActive(2,3,18176,1)
04-17 23:17:45 MainThread artisanlib.main      DEBUG  : message: Connected via MODBUS
04-17 23:17:45 Dummy-4    artisanlib.modbusport INFO   : invalidResult(1) => pymodbus error: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : Exception response
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(2,18176,3,False)
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(1,18176,3,False)
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(3,18176,3,False)
04-17 23:17:45 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:45 MainThread artisanlib.main      DEBUG  : message: Modbus Communication Error
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : readActiveRegisters()
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : clearReadingsCache()
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : readActive(2,3,18176,1)
04-17 23:17:46 Dummy-4    artisanlib.modbusport INFO   : invalidResult(1) => pymodbus error: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : Exception response
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : disconnect()
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : clearReadingsCache()
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(2,18176,3,False)
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(1,18176,3,False)
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : readSingleRegister(3,18176,3,False)
04-17 23:17:46 Dummy-4    artisanlib.modbusport DEBUG  : optimizer cache miss
04-17 23:17:46 MainThread artisanlib.main      DEBUG  : message: MODBUS disconnected
04-17 23:17:46 MainThread artisanlib.main      DEBUG  : message: Modbus Communication Error
04-17 23:17:47 MainThread artisanlib.main      INFO   : OFF MONITOR
04-17 23:17:47 MainThread artisanlib.main      DEBUG  : message: Scope stopped
04-17 23:17:47 MainThread artisanlib.modbusport DEBUG  : disconnect()
04-17 23:17:47 MainThread artisanlib.modbusport DEBUG  : clearReadingsCache()
04-17 23:17:47 MainThread artisanlib.s7port    DEBUG  : disconnect()
04-17 23:17:47 MainThread artisanlib.s7port    DEBUG  : clearReadingsCache()
04-17 23:17:50 MainThread artisanlib.main      INFO   : sendLog()
```

Durchstanzen, 2023-04-17:
Where did I find the log?

joeky888, 2023-04-17:
> Where did I find the log?

Hold ctrl+alt and click the PLUS icon.

Durchstanzen, 2023-04-17:
I have no email client setup… and honestly don’t want to…

Durchstanzen, 2023-04-17:
Ah ok… bad habit not to drink coffee!;)

MAKOMO, 2023-04-17:
On Windows you find the `artisan.log` file in the following directory:

`C:\Users\<USER>\AppData\Local\artisan-scope\Artisan`

Please also attach your `artisan-settings.aset.txt` file

Durchstanzen, 2023-04-17:
Thx I will have a look!

MAKOMO, 2023-04-18:
The Kapok roasting machines talk the rare ASCII dialect of MODBUS. The configuration API for this MODBUS dialog changed on updating the MODBUS lib underlying Artisan to v3.x in the Artisan v2.8.2 release, but the MODBUS client in Artisan was not updated accordingly. I just fixed this. Artisan v2.4.6 and v2.8.0 were still based on the MODBUS lib v2.x and communicate without any issues via MODBUS ASCII.

The fix is part of the current [continuous builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous). Please test if this fixes your issue and report back such that I can close this one.

ID-JerryChen, 2023-04-18:
I'm also having connectivity issues
Currently in operation roaster KAPOK 1.0

[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/11266114/artisan.log)
[artisanViewer.log](https://github.com/artisan-roaster-scope/artisan/files/11266117/artisanViewer.log)
[artisan_device.log](https://github.com/artisan-roaster-scope/artisan/files/11266120/artisan_device.log)

joeky888, 2023-04-19:
I can confirm this is no longer a problem, thank you.

MAKOMO, 2023-04-19:
Thanks for testing!

Durchstanzen, 2023-04-20:
Hey MAKOMO,
sorry to bother again but it didnt solve my problem... 

I attached the needed files.

[artisan.log](https://github.com/artisan-roaster-scope/artisan/files/11284890/artisan.log)

[Serial log.txt](https://github.com/artisan-roaster-scope/artisan/files/11284888/Serial.log.txt)

Durchstanzen, 2023-04-20:
and the settings

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/11284912/artisan-settings.txt)

Durchstanzen, 2023-04-20:
It doesn’t differ at all… I could do an other log with my other laptop connected running the older Artisan Version.
I simply upgraded… the arduino code is from an open source project.

MAKOMO, 2023-04-20:
Logging with the older Artisan is not productive as it does not produce a very informative log output, especially if there is no error.

A log from the Arduino side would be productive. Which open source project is that code from?

I just implemented quickly a server that is serving ET/BT to Artisan running under your configuration. That server receives exactly the same requests from Artisan v2.4.6 as it does from v2.8.2. The timing might be slightly different. You are sure that Artisan v2.4.6 is correctly communicating with your Arduino under those settings? The MODBUS serial comm port is correctly configured in both cases?

Device logging that you activated did not output debug logging of the MODBUS lib underlying Artisan. I did activated this now for the latest Continuous Build.

If you find the time, please download and install again the current continuous build. Ensure that under menu Config >> Device the "Logging" flag is ticked (as you had it in the settings you attached) and that the Artisan logging is set to DEBUG (as you had it on producing the log you attached). A short 30sec recording would be enough (ensure you have chosen the correct comm port in the MODBUS tab). Please attach again the resulting artisan.log file which should then contain raw MODBUS log messages. Thanks for your support!

Durchstanzen, 2023-04-20:
What shall I say… It does… I took picture before upgrading to 2.8.2 and checked the config afterwards… it’s the same…
Artisan says it connects with Modbus and then disconnects…

the arduino code is from here:
https://www.kaffee-netz.de/threads/kaffee-roesten-mit-severin-popcornmaschine-3751.80919/page-31

Post #608 

I will try and upgrade to your new version and try to post a log file.

MAKOMO, 2023-04-21:
Found an old Arduino UNO and install a small MODBUS RTU server for testing. Indeed the communication on Artisan v2.8.2 is never successfully established. The reason is that the first two requests from Artisan are not correctly answered by that server. The first response is empty and the second invalid. Only from the third request on the responses are correct. Those newer Artisan versions disconnect on the second transaction error which leads to the situation that the communication with the Arduino server never succeeds.

I increased the number of transaction errors before disconnecting and enabled also the pymodbus lib "retry on empty response" feature. Those changes are part of the latest continuous builds.

Could you please test and confirm that this fix works for you too?

Durchstanzen, 2023-04-21:
Will do so! And report. You do an amazing job by the way.

Durchstanzen, 2023-04-21:
Hey tested it with a temp probe, It shows one error connecting and then Reconnects with success. Showing both temps... So i asume it works.

Thanks a lot. Great work. Will test it with the complete setup, but dont have any doubt it will work.

MAKOMO, 2023-04-21:
Excellent! Thanks for testing and reporting. As said the first two responses are empty/invalid. There is nothing Artisan can do on this. This has to be fixed on the Arduino/MODBUS side. But after this initial hickup the communication seems to be stable. Thus I regard this as resolved and close the issue.

closed 2023-04-21 by MAKOMO

MAKOMO, 2023-05-12:
@Durchstanzen, @joeky888, @ID-JerryChen: we've just uploaded a (signed) beta of the next Artisan release for review by Artisan experts like you. 

https://github.com/artisan-roaster-scope/artisan/releases/tag/v2.8.3-beta

This version includes only a few minor new features, but lots of bug fixes. Release History:
https://github.com/artisan-roaster-scope/artisan/blob/master/wiki/ReleaseHistory.md

Maybe you find time to test this version on your setup and report back issues.

Thanks,
Marko & Dave

joeky888, 2023-05-14:
I can confirm there is no issue with the version v2.8.3-beta.

MAKOMO, 2023-05-14:
@joeky888 : thanks for testing and reporting!

Durchstanzen, 2025-06-20:
Hey Marko,It’s bean a while since I had this connection issue with my homebuild roaster. Last known working version of was 2.4.4 for me and with 3.1.4 I’m getting the same problems as it was back then. It connects and then Disconnects. A bit later it starts the circle again.If I remember correctly this was due to a to slow response by my „system“. Is there a chance you can fix that again?Greets, D.Am 14.05.23 um 19:24 schrieb Marko Luther
                
            
            
                
                    Von: "Marko Luther" ***@***.***>Datum: 14. Mai 2023An: "artisan-roaster-scope/artisan" ***@***.***>Cc: "Durchstanzen" ***@***.***>,"Mention" ***@***.***>Betreff: Re: [artisan-roaster-scope/artisan] Modbus com error reopen please (Issue #1145)
                
                
@joeky888 : thanks for testing and reporting!

MAKOMO, 2025-07-07:
What happens if you set the Serial Delay parameter to 5sec (default is 0.5sec) in menu Config >> Ports, MODBUS tab in current Artisan v3.2?