artisan-roaster-scope/artisan #1147: Troubleshooting Issue

joe2k1, 2023-04-13:
I have the following error message on my Acer Aspire V 14 Touch laptop. The software is not working after an operating system upgrade. I get the following message under Artisan.

Exception : PhidgetManager couldn’t be started. Verify that the Phidget driver is correctly installed. 

I am running 5.15-0.69 generic #76 Ubuntu SMP - March 17 2023. 

Let me know if you need any other information as I would love to get roasting again.

Thanks in advance.

CarefreeBuzzBuzz, 2023-04-14:
> Sorry about that!
>
> Describe the bug
> The software worked before an operating system update.
>
> To Reproduce
> You get the error message whenever you try to run the Artisan software.
>
> Expected behavior
> The software use to work flawlessly prior to operating system upgrade/update.
>
> Screenshots
> See comment about.
>
> Setup (please complete the following information):
>
> Artisan Version [e.g. v2.4.4]:
> Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec <https://github.com/artisan-roaster-scope/artisan/commit/97e6dec3b59277aff2b37afa6035b690a4d1c499>]:
> Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:
> Connected devices or roasting machine [e.g. Probatone 5]:
> See above again it was all working perfectly prior to the update.
> Additional context
> Add any other context about the problem here.
>
> Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
> Please attach any relevant Artisan *.alog profiles.
>
> Note that you need either add a .txt extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading.

You might have to go into the Security settings and approve the software.  But please update to the latest version.  Save your settings under the Help menu first.

joe2k1, 2023-04-14:
> You might have to go into the Security settings and approve the software. But please update to the latest version. Save your settings under the Help menu first.
> […](#)

Hi, Thanks for your help. I can confirm that the software is running the latest version. Can you be more specific about security settings and approving the software. It is very clear that something is wrong with compiling the Phidget driver.

MAKOMO, 2023-04-14:
This is a very confusing report. You edited your initial message such that the replies to it do not make sense anymore.

Obviously you have a problem with your Phidget driver. This is not part of Artisan. Contact Phidget Support on how to proper install their driver on your machine. Without that driver correctly installed Artisan will not be able to connectto your Phidgets.

closed 2023-04-14 by MAKOMO

joe2k1, 2023-04-14:
Just as a followup, everyone was correct it was an issue of the Phidget driver/library not being installed at all. I am running Linux Mint. I should have mentioned that part earlier. I worked with the technical support department to determine that i had to run these 2 very specific command lines to get Artisan working again. 

curl -fsSL https://www.phidgets.com/downloads/setup_linux | sudo -E bash -

sudo apt-get install -y libphidget22

Once I ran those two lines separately, I was able to get Aristan working again, no more error messages. 

Thanks again to everyone and I am leaving this here to help others who may run into this very strange off issue after an operating system upgrade.