artisan-roaster-scope/artisan #1160: It seems wayland is crashing on Ubuntu 22.04 (on virtualbox) after certain actions in Artisan

lookingafar, 2023-05-08:
**Describe the bug**
It seems wayland is crashing on Ubuntu 22.04 (on virtualbox) after certain actions in Artisan. I managed to reproduce it with a roast properties window, but I saw crashes after some other actions as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Install continuous build on Ubuntu 22.04 (mine is running on Wayland in Virtualbox)
2. launch artisan
3. press CTRL+T to open roast properties window
4. Click on the down arrow after title to select roast title
5. Wayland crashes, causing ubuntu to relaunch the desktop

**Expected behavior**
Not sure if it's Ubuntu problem or Artisan problem, but it seems that Artisan is causing the crash.

**Screenshots**
I attached a video

**Setup (please complete the following information):**
 - Artisan Version 2.8.3 (continuous build)
 - Artisan Build (number in brackets shown in the about box) 5be6ec5
 - Ubuntu 22.04, running on Wayland in Virtualbox
 - None, the crash also happens after factory reset.

**Additional context**
Please watch the video

https://user-images.githubusercontent.com/36126988/236900741-443f3438-0061-4f32-8549-d77f7494da01.mp4

MAKOMO, 2023-05-09:
Thanks for the additional testing and this report! I did run that build on a Ubuntu 23.04 (on a macOS host) under Wayland for a while (including the actions you show in the attached video) and couldn't get it to crash. Could it be a low memory issue on your VM or an issue specific to Ubuntu 22.04? The Qt 6.5 version underlying this build officially supports still 22.04.

lookingafar, 2023-05-15:
It seems to be an issue specific to Ubuntu 22.04.  The crash is reproducible with a fresh 22.04 installation with 8gb or RAM, but on 23.04 Artisan runs without problems.

MAKOMO, 2023-05-15:
Thanks for this further investigation. So I will close this here as it seems to be a platform issue which we unlikely can resolve on the Artisan level.

closed 2023-05-15 by MAKOMO