artisan-roaster-scope/artisan #1170: Ver 3.8.3 beta and Kaleido M10

MarcelPr, 2023-05-18:
These are a few problems with the 3.8.3 beta version. 
I have a Kaleido M10 roaster, and I click on
Config>Machine>Kaleido....

1) If I install the Serial version, it does not see my roaster. 
However, the legacy version connects to my M10 with no problems.
Same port COM7, same baud rate 9600.
By the way, I have a Surface Book 2 laptop running WIndows 11, so
I probably need a Serial version.

2) I added an LCD for SV via
Devices>Extra Devices  and checking the corresponding box.
However, the LCD and the slider for SV show different numbers. 
The difference is either 0 or 1 or 2, depending on the value.

3) The drum number in the slider cannot go lower than 10, even 
     though the drum is not rolling. 

4) A few minor suggestions. It took me some time and a few ruined 
    batches to change the settings as follows. I guess it should be the default.
     - Have Auto Charge checked by default
     - For the manual roast set up an event to increase the SV to the maximum
        after the Charge or TP.
      - In the PID settings change the SV to manual, right now it is for background.    
         Cannot change it when roasting.
      - Most important, in the background properties uncheck the box Playback Events.
         In the manual mode it changes the power which is not desirable.
       
Thank you!  
         

**Setup (please complete the following information):**
 - Artisan Version : 2.8.3 beta
 - Artisan Build : 302963a
 - Platform: Windows 11
 - Connected devices or roasting machine: Kaleido M10

MAKOMO, 2023-05-27:
> These are a few problems with the 3.8.3 beta version. I have a Kaleido M10 roaster, and I click on Config>Machine>Kaleido....
> 

The Kaleido support is not yet final.

> 1. If I install the Serial version, it does not see my roaster.

That one is for a newer version of the Kaleido protocol still to be released for new machines.

>    However, the legacy version connects to my M10 with no problems.

This one is for currently available models.

>    Same port COM7, same baud rate 9600.
>    By the way, I have a Surface Book 2 laptop running WIndows 11, so
>    I probably need a Serial version.

I cannot follow this reasoning. Sorry.

> 2. I added an LCD for SV via
>    Devices>Extra Devices  and checking the corresponding box.
>    However, the LCD and the slider for SV show different numbers.
>    The difference is either 0 or 1 or 2, depending on the value.

A delay caused by communication. Absolutely to be expected at 9600 baud and sampling rate of 2sec.

> 3. The drum number in the slider cannot go lower than 10, even
>    though the drum is not rolling.

On purpose

> 4. A few minor suggestions. It took me some time and a few ruined
>    batches to change the settings as follows. I guess it should be the default.
>    
>    * Have Auto Charge checked by default

AutoCHARGE is checked by default. The Kaleido machine setup does not touches this. So you must have unticked it yourself.

>    * For the manual roast set up an event to increase the SV to the maximum
>      after the Charge or TP.

You can define an alarm rule to achieve this.

>    * In the PID settings change the SV to manual, right now it is for background.
>      Cannot change it when roasting.

This is on purpose by Kaleido. The PID mode can be changed while roasting. If PID is on they assume the SV is coming from the background profile. Just change this if you like. You can also define custom event buttons with some Artisan Commands to switch between the different PID SV modes.

>    * Most important, in the background properties uncheck the box Playback Events.
>      In the manual mode it changes the power which is not desirable.

Same as for AutoCHARGE, the Kaleido machine setups don't touch the background playback settings. The Artisan factory default for this is OFF. You must have ticked this one yourself at some point.

> 
> Thank you!

I'll close this as there is nothing to be changed.

> 
> **Setup (please complete the following information):**
> 
> * Artisan Version : 2.8.3 beta
> * Artisan Build : [302963a](https://github.com/artisan-roaster-scope/artisan/commit/302963acd73f0b70250323ec9257bc2f100e3e8f)
> * Platform: Windows 11
> * Connected devices or roasting machine: Kaleido M10

closed 2023-05-27 by MAKOMO

MarcelPr, 2023-06-01:
>    1. I added an LCD for SV via
>    1. The drum number in the slider cannot go lower than 10, even

I understand all the comments, except for one related to the different SV
values in the lcd and the slider.  They are different not for some small
period of time, but constantly. Say, you set the slider to 380 F and the
lcd shows 378 forever. Is it  normal?