artisan-roaster-scope/artisan #1175: BT RoR smoothness change in beta testing

Brewzologist, 2023-05-22:
**Describe the bug**
!! THIS MAY NOT BE A BUG !!  Testing beta v2.8.3 and reporting in case it might be code related, but could be my BT probe/hardware. Understand if there is no resolution here and will continue to look into my hardware.

**To Reproduce**
Steps to reproduce the behavior:
1. I performed 4 roasts using the v2.8.3 beta build
2. I noticed unusual response in BT RoR in each of the 4 roasts where the line is less smooth than prior Artisan versions.  I use Phidgets with RTD probes.
3. No changes to my hardware or ASET file have occurred.

**Expected behavior**
I expect my BT RoR line to have similar smoothness as in prior releases of Artisan.  I have not noticed this before but acknowledge it could be a new hardware issue with my BT probe perhaps.

**Screenshots**
See the BT RoR line in the attached examples from a v2.8.2 roast and the v2.8.3 beta roast.  This same behavior occurred in 3 additional roasts done on the same day.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.3 beta
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: 66f5693
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: Windows 10 with all updates installed
 - Connected devices or roasting machine [e.g. Probatone 5]: Huky 500T using VINT Phidgets with RTD probes, and Arduino custom fan/gas control

**Additional context**
None

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[Possible beta bug.zip](https://github.com/artisan-roaster-scope/artisan/files/11531606/Possible.beta.bug.zip)

Brewzologist, 2023-05-22:
FYI;  After this occurred, I reinstalled v2.8.2 and loaded the roast profiles with the less-smooth BT RoR but they didn't display any differently than with v2.8.3 beta.

MAKOMO, 2023-05-27:
There is a difference between 2.8.2 and the 2.8.3 beta. The beta adds supports for the new TMP1200_1 module which comes with a lower maximal data rate. I move things around such that the original TMP1200 can still use its highest data rate. If this is at all an issue (difficult to test) it should be fixed by Commit 61febf0a906f72878939e134f61b3ab517c5edab and will be part of the Release Candidate to be published soon.

closed 2023-06-06 by MAKOMO

Brewzologist, 2023-06-27:
FYI;  I no longer notice this issue in v2.8.4 release.

roasterdave, 2023-06-27:
Thank you @Brewzologist for the confirmation.