artisan-roaster-scope/artisan #1179: Artisan+ No Graphs displayed under the "Roasts" page

bigl00z3, 2023-05-23:
**Describe the bug**
A clear and concise description of what the bug is.

In Artisan+, under the [Roast page](https://artisan.plus/en/roasts) , graphs should be visible in the lower part of the page. At the present time, the following message is shown instead of the graphs `Graphs of your data are shown here as soon as there is enough data.` even when they are hundreds of roasts available.

[image]

Graphs on the[ Stores page](https://artisan.plus/en/stores) are displayed as usual.

Nico

MAKOMO, 2023-05-23:
Dear Nico,

thank you very much for reporting this issue timely. A wrong deploy messed up things a little bit. Should be fixed be now.

Best,
Marko

closed 2023-05-23 by MAKOMO

bigl00z3, 2023-05-23:
I can confirm the fix.