artisan-roaster-scope/artisan #1180: Artisan+: Roast list order issue

bigl00z3, 2023-05-23:
When looking at the fix for [issue 1179](https://github.com/artisan-roaster-scope/artisan/issues/1179) I think I found an other one:

In Artisan+ [Roast page](https://artisan.plus/en/roasts), it should be possible to change the order of the roast by label and invert the order by clicking on the button with the arrow up ↑ or down ↓ 

[image]

Changing by label (Roast date, ID, Label, Batch number...) works but not inverting the order by clicking a seond time.

Nico

paulholleis, 2023-05-24:
In fact, the initial sort is wrong. Therefore, clicking once on the "roast date" button does nothing. Clicking twice will invert the sort order. Thanks for reporting, this will be fixed shortly!

MAKOMO, 2023-05-27:
Fixed

closed 2023-05-27 by MAKOMO