artisan-roaster-scope/artisan #1188: individual Keyboard Shortcuts for Event Buttons

peterweiser, 2023-06-01:
**Describe the solution you'd like**
It would be very useful to have a function which makes it possible to add an individual keyboard shortcut to an event button.

roasterdave, 2023-06-03:
There are several existing shortcuts that can make event entry from the keyboard. There is no current plan to add more.  Please read on to understand what is available.

For the roast events use the SPACE BAR to mark a roast event and move the next.  Enable and disable this feature by hitting the RETURN/ENTER key.  Once enabled the event to be marked is highlighted, like the CHARGE button in the image below.  Tapping the SPACE BAR will mark CHARGE and move the highlight to DRY END.  Tap SPACE BAR again and DRY END will be marked and the highlight moved to FC START.  Use the LEFT and RIGHT ARROW keys to manually move the highlight.  When using this mode the buttons are still available to click with the mouse.  Auto CHARGE and Auto DROP still work as normal. 

There are shortcuts for the Special Events too.  Many users create custom event buttons to enter values for the Special Events.  There are keyboard shortcuts for this too.  The keys q,w,e, and r correspond to special events 1, 2, 3 and 4. In a default configuration these are (1)Air, (2)Drum, (3)Damper and (4)Burner.  Normally a two digit numeric value must follow the shortcut letter to set the value.  For example, 'q75' means Air=75, and 'r20' means Burner=20.  If the corresponding event slider has a max value of 101 or greater then three digits must follow the the shortcut letter.  In this case values less than 100 require a leading zero to make them three digits, e.g. 'q075'.  Note that it is not necessary for event sliders to be enabled for these shortcuts to work.

If you use custom event buttons for any purpose it is possible to trigger any of them from the keyboard.  You need to know the custom event button's number (Config>> Events>> Buttons).  From the keyboard hit Ctrl+Shift+B (Windows) or OPTION-B (Mac) followed by the two digit button number, e.g. a '07' will fire button number 7. Custom event buttons are quite powerful.  You could even use them to fire one of the roast event buttons.  And much more.  Coupled with a keystroke macro program on you computer you can create your own shortcuts to do almost anything in Artisan.  You could even use an external button keypad to send keystrokes to Artisan.

[image]

bravowren, 2023-06-04:
One of the keyboard shortcuts I discovered purely by accident (that I have not seen documented) is TAB followed by UP/DN arrows. On a Mac, TAB moves through Sliders in sequence, and the arrow keys increment the value of the selected slider. This is extremely useful if you have disabled certain local controls in favor of Artisan control - adding a mini-keyboard  with programmable keys gives you control back at the roaster.

peterweiser, 2023-06-06:
Thank you for your detailed answers - I couldn´t find that information somewhere. I ve already tried out these solutions and they work great for my usecases.

commit 017aa69e (2023-06-06): - enhance keyboard shortcuts help file (resolves #1188)

- disable tab to sliders when keyboard shortcuts is active at roast start.
- allow only single quotes and unix line endings in output of xlsx_to_artisan_help.py
closed 2023-06-06 by MAKOMO