artisan-roaster-scope/artisan #1191: alog file temperature converter?

MarcelPr, 2023-06-02:
Some people share with me their .alog files, but they use C, and I am using F. Is there any way to convert a saved .alog file from C to F? If not, this would be a very convenient new feature, I guess. Convert not only the BT, ET, but also the RoR, SV, everything. So that the converted .alog file can be used as a background.

Thank you!

CarefreeBuzzBuzz, 2023-06-02:
If you look at the QuickStart guide, you will see how to do it___________________________Michael Herbert, CarefreeBuzzBuzz

MarcelPr, 2023-06-02:
Thank you! Perhaps I am missing something. Yes, I know that there is a config>temperature option. But when I load a C background profile, and after that I change the temp from C to F, the numbers for the BT stay the same. Say, it was 180 C, and after changing the temp it is now 180 F. So, I cannot use the profile for the background.

Am I wrong? Is there something else in the QuickStart Guide?  Sorry for my ignorance.

roasterdave, 2023-06-03:
First set Artisan to the desired display mode, **Fahrenheit Mode** or **Celsius Mode** using **Config>> Temperature**.  Then open the profile and/or background profile you want to view.  The temperatures are automatically converted when the profile is opened.  Note that any profiles open when the mode is changed will need to be reopened because the temperature conversion happens while opening the profile.  

If you want to permanently save a converted profile there are three ways to do it.  
**File>> Save As** works when viewing a profile as described above.  
**Tools>> Convert Profile Temperature** will perform the conversion immediately and then prompt if you want to Save when you close the profile.  
**File>> Convert To** allows you to choose one or multiple files to convert in bulk without displaying them.

MarcelPr, 2023-06-04:
Thank you so much for the very detailed instructions!

closed 2023-06-06 by MAKOMO