artisan-roaster-scope/artisan #1198: sliders are interchanged

peterweiser, 2023-06-10:
**Describe the bug**
The sliders are somehow interchanged.

**To Reproduce**
Steps to reproduce the behavior:
Since the update to 2.8.3rc, the sliders are interchanged somehow. The slider for the air fan now controls the drum speed and the slider for the burner controls the air fan. The drum slider controls the burner. So the names of the sliders doesn´t correspond to the funktions anymore.
I tried to interchange the modbus commands back, but found out that then also the alarm buttons work wrong.
The behaviour is the same in the latest Artisan build.

**Setup (please complete the following information):**
 - Artisan Version [v2.8.3]:
 - Artisan Build [fc2da5c:]
 - Platform (Windows + 10 22H2:
 - Connected devices or roasting machine [Dätgen DR12 + modbus + Phidgets]:

**Additional context**
Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.

[artisan-einstellungen_6b_2023.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/11712211/artisan-einstellungen_6b_2023.aset.txt)

MAKOMO, 2023-06-10:
In the settings file you attached the event names (menu Config >> Events, 1st tab) are set to the Artisan defaults (Air, Drum, Damper, Burner). Most likely you had the names set to (Drum, Burner, Damper, Air), why ever. It remains unclear how those event names got changed. Could have happened by loading profiles from other setups, but shouldn't.

The simplest is to change the event names back to your custom order. That way the profiles recorded will fit to your previous ones.

Note that you are running a heavily customised setup.  Which in part might generate problems. For example the Quantifier setup (4th tab in Events dialog) is most likely troublesome. All quantifiers are active observing the BT (or ET in one case), thus curves which for sure do not hold any event related data. Also the min/max limits do not correspond to those in the Sliders tab. I strongly suggest to completely disable all 4 quantifiers.

I will keep this open, but without more information on when and how the event names got swapped I fear this cannot be resolved.

MAKOMO, 2023-06-30:
Stall

closed 2023-06-30 by MAKOMO