artisan-roaster-scope/artisan #1207: After update to 2.8.4 via rpm, can't run as a regular user - only root

DmitryBurstein, 2023-06-22:
Updated the program to its latest version with dnf on Fedora 38 machine. Since then it can only run as super user, not as a regular one. No error code displayed.

MAKOMO, 2023-06-30:
Is this related to Issue #1001?

MAKOMO, 2023-06-30:
What happens if you start it (without sudo) like this

```
# /usr/share/artisan/artisan
```

Do you see any error output?

DmitryBurstein, 2023-07-01:
It probably is related to #1001. I have no problem starting it as a super user - with no output on the command line. As a regular user however it doesn't start. If I just run "artisan" as a command, no output displayed. When I tried the full pass "/usr/share/artisan/artisan", I got the following:
/usr/share/artisan/artisan: symbol lookup error: /usr/share/artisan/libQt6WaylandClient.so.6: undefined symbol: wl_proxy_marshal_flags

DmitryBurstein, 2023-07-01:
I'm not sure it's related, but besides artisan's own `libQt6WaylandClient.so.6`, and a system-wide one, both of which do have `wl_proxy_marshal_flags` exported, there is another `libQt6WaylandClient.so.6` on my machine - in the /opt/ path, and this one does not export `wl_proxy_marshal_flags`. Could it be when artisan runs under a regular user it somehow tries to use the wrong library?
On the other hand, when I moved the "wrong" library away, and ran artisan again, I got the same error message as before - so I'm not sure whether it helps...

MAKOMO, 2023-07-01:
I have not much knowledge on the Linux platform, never used Linux myself. Hope that someone will chime in with some expertise that can help to resolve this.

DmitryBurstein, 2023-07-02:
As a quick and dirty hack, I found that adding `WAYLAND_DISPLAY=''` to your `exec` line in `/usr/bin/artisan` (or just running `WAYLAND_DISPLAY='' /usr/share/artisan/artisan`) does solve the problem. However I'm not sure this is the proper way to do it or will not cause some other problem.
In my native environment `WAYLAND_DISPLAY=wayland-0`

virtuosonic, 2023-09-20:
In Linux in order to know which libraries are loaded at runtime by an executable you can use the ldd command like this
```
$ldd /bin/ls
```
You can then run as sudo to see if there's a difference

MAKOMO, 2023-11-20:
Seems to be a Linux configuration issue

closed 2023-11-20 by MAKOMO