artisan-roaster-scope/artisan #1214: Enhancement - Y axis F and C at same time

ArmedLeftist, 2023-06-29:
My apologies if this is already an existing request!

It would be helpful to me if the Y axis of the roast graph could show both  F and C temps at the same time. I unfortunately grew up using the Farenheit system (I know I know), and I still struggle making quick, reflexive conversions when experimenting with a new profile I’ve found, that is based on Celsius. (Plus, just “geeking out” online reading or listening to podcasts, watching YTs, etc., where base reference is to C, would become more meaningful to me when it comes to trying thier approach at roast time. Thanks very much.

ArmedLeftist, 2023-06-30:
So much for the definition of Quality being “delight your customer”. 

Dave…given that you are a simple “contributor” to an open-source and collaborative development effort, do you really have the authority to summarily dismiss an enhancement request; particularly in the aggressive and counter-productive manner that you did  (then weakly transitioning to a patronizing passive-aggressive behavior style).

You had no way of knowing, but I am a Sr Engineering Fellow (with 40 yrs experience), for one of the most advanced technological companies on the planet, I have fired hundreds of junior engineers like you. Perhaps you can now grok the absurdity of the tone of your reply. I assure you that you are WAY out of line “speaking” to me (or anyone for that matter), like that. 

Returning to the question at hand and considering for example, functionality that exists in MS  Outlook, why do you think there is an ability to display two time zones in the calendar function Dave? Because it relieves the user from having to do are the juggling you suggested (demanded) in your response. Clearly, users asked for it (or, like this, the blue of such a feature was so intuitively obvious, it didn’t even need to come in as an enhancement request. 

Any other contributors feel like considering this request constructively, or does Dave own this effort and therefore he does the Deciding for everyone?

CarefreeBuzzBuzz, 2023-06-30:
The request is interesting.  If you are a senior engineer please rally resources and create a pull request.  There are no resources at Artisan to do this.  Let us know how your efforts go.  Thanks. Michael Herbert CarefreeBuzzBuzz

ArmedLeftist, 2023-06-30:
@CarefreeBuzzBuzz Thank you but unfortunately, my area of engineering is not software engineering.

MAKOMO, 2023-06-30:
> So much for the definition of Quality being “delight your customer”.

You are not a customer as this is not a commercial product.

> 
> Dave…given that you are a simple “contributor” to an open-source and collaborative development effort, do you really have the authority to summarily dismiss an enhancement request; 

Yes, he has as one of the two main code contributors spending countless hours on this project for free!

> particularly in the aggressive and counter-productive manner that you did (then weakly transitioning to a patronizing passive-aggressive behavior style).

I cannot see anything aggressive nor passive-aggressive in his answer. It was short and to the point and included hints on how to work around.

> You had no way of knowing, but I am a Sr Engineering Fellow (with 40 yrs experience), for one of the most advanced technological companies on the planet, I have fired hundreds of junior engineers [...]

The background of people posting here does not matter at all nor does our background. Here everything is about the Artisan project.

> Returning to the question at hand and considering for example, functionality that exists in MS Outlook, why do you think there is an ability to display two time zones in the calendar function, Dave? Because it relieves the user from having to do are the juggling you suggested (demanded) in your response. Clearly, users asked for it (or, like this, the value of such a simple to implement feature was so intuitively obvious, it didn’t even need to come in as an enhancement request.

As Michael, just submit a PR with your simple implementation or alternatively just fork the project and develop it in the direction you like. This is open-source after all. If you do not have the technical competence to do so, you can always hire someone to do the work on your behalf.

> 
> Any other contributors feel like considering this request constructively, or does Dave own this effort and therefore he does the Deciding for everyone?

He own this effort and he is well authorised to decide on behalf of the the Artisan team. Dave considered this request constructively and reported you the outcome timely. 

Please follow the [GitHub Community Guidelines](https://docs.github.com/en/site-policy/github-terms/github-community-guidelines#hate-speech-and-discrimination) and be respectful, especially to those who offer the world their talent for free!

ArmedLeftist, 2023-06-30:
Thanks Marko. It’s all coming clear to me now. I now have a much better understanding of the existence of the “feature request” function provided.As to Dave being one of the main contributors, thanks for that as it offers me the opportunity to take a bit of a tangent here to compliment him (and you and…whomever), on his/your work on this project. Artisan is a really great and helpful application! I love using it. Still…just….”Not gonna happen (period)”. Geez.Whatever.

CarefreeBuzzBuzz, 2023-06-30:
Here’s another idea.  

You say that software isn’t your area.  

IF you really, really want this implemented?  YES, are you sure?  Ok Yes, here is how to do it.  

Hire Marko as a consultant <https://shop.artisan.plus/shop/en/consulting/>.  BOOM Done.  Click the link.  

While I understand your request.  As you have now learned a bit, Artisan isn’t an enterprise.  It’s the contributions of three core people to the coffee community.  Only so much can be done. So you can choose between a cheat sheet and getting it done.  

Hope this helps.  

Happy Roasting.  

----------------------------------------
Michael Herbert, Head Roaster
Order at www.carefreebuzzbuzz.com <http://www.carefreebuzzbuzz.com/>
Support Artisan RoasterScope  <https://artisan-scope.org/>by using 
Artisan.Plus <https://artisan.plus/en/>

ArmedLeftist, 2023-06-30:
Thanks. I may very well do that although in the interim I have contacted a friend who is a Sr Software Engineer in my company to see if he might be interested/available when he returns from family vacation. Thanks again.