artisan-roaster-scope/artisan #1221: Artisan crash report: 2.8.4 on MacOS 12.6

fatrabbit-la, 2023-07-06:
Artisan recently crashed on me twice when I was trying to change the labels for two existing buttons. I was able to access the 'Buttons' window and make the text and value changes but when I clicked to save the changes, the app crashed. I think, but don't remember clearly, that the app was 'ON' when I did this because I was in between batches. I seem to recall turning it 'OFF' when trying a third time and then it was fine.

MAKOMO, 2023-07-07:
Thanks for your report. To fix this we need a way to reproduce this. Your Artisan settings with an exact protocol of steps to reproduce this should do.

fatrabbit-la, 2023-07-07:
Settings attached. Steps I took that lead to crash…

Config > Events > Buttons…

Change LABEL and VALUE for two existing Fan buttons. Click OK.

Crash.

As mentioned previously, I think the software was ‘ON’ when this happened but I don’t clearly remember. I just tried it both ‘ON’ and ‘OFF’ and it did not crash but I’m also not connected to my roaster right now like I was before.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

MAKOMO, 2023-07-08:
Nothing attached. You need to drag-drop the file to a comment box in a Browser window here on Github, an email reply will not work.

fatrabbit-la, 2023-07-08:
[artisan-settings_COFFEE copy.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/11992250/artisan-settings_COFFEE.copy.aset.zip)

Here is the file, hopefully. I got an error when dragging and dropping that the file type wasn't supported so I had to zip it in order for the Comment box to accept it. Am I doing something wrong just trying to attach the file?

MAKOMO, 2023-07-09:
> [artisan-settings_COFFEE copy.aset.zip](https://github.com/artisan-roaster-scope/artisan/files/11992250/artisan-settings_COFFEE.copy.aset.zip)
> 
> Here is the file, hopefully.

Thanks! Now it is attached.

> I got an error when dragging and dropping that the file type wasn't supported so I had to zip it in order for the Comment box to accept it. Am I doing something wrong just trying to attach the file?

We have the following note in our Issue Template (which you obviously deleted with all the rest of the information requests we have for issues):

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._

On the subject: I connected my 1048 and tested v2.8.4 with you setup at a 0.25s sampling frequency and at your 3sec sampling frequency and could not reproduce this crasher on my mac. I tested while OFF, ON and START.

fatrabbit-la, 2023-07-18:
Continuation of this issue but tonight was different. I roasted a total of 6 batches. For the first four batches, when I clicked 'OFF' to end the roast, the typical dialog came up to 'save, cancel, don't save'. I clicked 'Don't Save', as I often do, and the app crashed right then and there.

I restarted the machine fresh and the same thing happened the next 3 roasts. Thankfully, no issues while roasting. I then expected it to crash for the last two roasts but no issues and I was able to continue uninterrupted.

Nothing about my system has changed. I've been using the same computer and OS for years (other than minor updates within the existing OS12.
[CrashReport.pdf](https://github.com/artisan-roaster-scope/artisan/files/12075647/CrashReport.pdf)

Apple crash report attached. I also shot video of the issue but it's just me clicking 'Don't Save' and then the app crashes.

Artisan has been extremely stable for me throughout the years. So far as I can remember, I've never had the app actually crash on me so this is a new one.

MAKOMO, 2023-07-18:
This is a good further data point. The CrashReport.pdf is as usual not very conclusive as it reports on things deep in the libs we are using, but not our code and on crashs our Artisan log files don't get the chance to document anything. Therefore the cause of crashs are very hard to understand and usually the consequence of some races on resources by parallel threads. In this case there seems to be a resource issue (maybe timing) involved as it does not happen always, but infrequently. I will review the relevant code and see if I can get a better guess on what is happening here. Thanks for this additional data.

MAKOMO, 2023-07-19:
I received a second report (sadly not via our issue tracker) on crash of Artisan v2.8.4 on pressing OFF on macOS 12.x (M1) using Phidgets. Do you have the latest Phidgets driver installed? Is your Phidget module running the latest firmware version? Is there a specific reason you are still on macOS 12 and did not upgrade to macOS 13? The issue seems to be related to this Intel emulator Rosetta and macOS is running a rather old version of it.

Jay-NCNY, 2023-07-19:
I too have had 2.8.4 crash on my M1 MacBook Pro running MacOS 13.4. Both times, I was actively monitoring temperature while between batches, but I do not record these values so have no log file. I have attached two crash reports on the off chance they’ll be helpful (-:

In both cases, I had already:
Warmed up pre roast (monitor temp without logging)
finished roast 1 (logs look fine in both cases)
started my between batch protocol (monitor temp without logging)

I looked away for a few seconds, maybe as much as 2 minutes, and found it had crashed.

Both times, I finished my roasting with a copy of 2.8.2 I have installed separately.
On the off chance I might have installed a corrupted image, I downloaded and installed anew after the first crash, but it crashed again the next time I ran it.

**Setup**
 - Artisan Version 2.8.4
 - Artisan Build ac17fbe
 - Platform macOS 13.4
 - Connected devices or roasting machine: Arduino Leonardo designed to look like a TC4

Hope this helps!

[Artisan-2023-07-13-004405.txt](https://github.com/artisan-roaster-scope/artisan/files/12100811/Artisan-2023-07-13-004405.txt)

[Artisan-2023-07-09-022035.txt](https://github.com/artisan-roaster-scope/artisan/files/12100810/Artisan-2023-07-09-022035.txt)

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12100809/artisan-settings.txt)

fatrabbit-la, 2023-07-22:
Sorry for the late reply. The driver I have is Phidget21 - 2.1.8 which appears to not be the latest. I see there’s a Phidget22.

The reason I did not upgrade to macOS 13 is due to frustrating issues I’ve run into in the past when “upgrading” to a latest OS. Typically, if things are working on my computer, I leave them be, only doing the minor upgrades within a given OS unless there’s a compelling reason to upgrade to the supposed latest and greatest.

But I’m open to updating if highly recommended.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

MAKOMO, 2023-07-23:
> Sorry for the late reply. The driver I have is Phidget21 - 2.1.8 which appears to not be the latest. I see there’s a Phidget22. 

This is not possible. Artisan 2.x does not support the Phidget21 driver and only works with the Phidget22 driver. You must have both drivers of very different generations installed. This is not a problem as Artisan is just ignoring the Phidget21 driver. However, for the Phidget22 drivers there is an update about every few weeks and those updates are important to keep your Phidgets system running stable. Update is trivial just download and install the latest driver from phidgets.com. You should also update the firmware of your modules after driver installation. The Phidgets Control Panel app will show the modules in red if there is a new firmware. Just click on them to update.

> The reason I did not upgrade to macOS 13 is due to frustrating issues I’ve run into in the past when “upgrading” to a latest OS. Typically, if things are working on my computer, I leave them be, [...]

There are now reports for crashes of Artisan v2.8.4 on macOS 13 (M1) also so a system update might not help you to prevent those. The crashes seem also to be independent of the Phidgets driver and the `say` alarms. It is still very unclear what is responsible for the crashes, but they seem to occur only on M1 processors.

Dan Fat Rabbit Coffee ***@***.*** ***@***.***> [www.fatrabbitcoffee.com](http://www.fatrabbitcoffee.com) <http://www.fatrabbitcoffee.com/>

fatrabbit-la, 2023-08-15:
Any updates regarding this issue? Just roasted again for the first time in a while and Artisan crashed after my first roast. As in the past, thankfully it didn’t crash until after I finished and clicked the ‘Don’t Save’ button in the dialog box.

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

MAKOMO, 2023-08-20:
We still have not any idea what is going wrong for v2.8.4 on those M1 macs and we did not get a clear response if the previous Artisan releases perform any better. This is mostly due to our inability to reproduce this issue for the lack of hardware. Donations for this project are at such an all-time low right now that it is unlikely that we get our hands on an M1 mac anytime soon.

What we did is to change the OFF action handling as this was reported (Issue #1235) to cause crashes rather often. Not sure if this makes a different at all. The issue might also be caused by a build problem and it seems that M1 macs are more sensitive to those than Intel macs. Also issues in the underlying Qt GUI lib might be the cause. Thus we produced a build with the above OFF handling change, based on an updated Qt lib and build with a somewhat updated build infrastructure.

https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous

Maybe someone of you finds the time to test this on their M1 mac and reports back success or failure. Note that this build is not signed nor notarized by Apple, thus you need to start it the first time via a right-click to the app icon selecting OPEN and accept the conditions.

fatrabbit-la, 2023-08-20:
I’ve had the same M1 laptop for at least a year and I want to say even longer than that (my memory for these things is crap) and have never had a crashing problem until this latest Artisan release. I haven’t tried going back to a previous version but maybe that’s the way to go until things get ironed out. But I’m also happy to first try this modified build and will report back soon as I roast again.

Dan

Fat Rabbit Coffee
***@***.***
www.fatrabbitcoffee.com

MAKOMO, 2023-08-21:
> I’ve had the same M1 laptop for at least a year and I want to say even longer than that (my memory for these things is crap) and have never had a crashing problem until this latest Artisan release. 

Thanks for this confirmation that this is a v2.8.4 only issue

> I haven’t tried going back to a previous version but maybe that’s the way to go until things get ironed out. But I’m also happy to first try this modified build and will report back soon as I roast again. 

Awaiting your report!

fatrabbit-la, 2023-08-21:
Ran Artisan 2.8.4 NOT connected to the roaster and Phidget - just the app by itself. It did not crash.

Ran Artisan 2.8.4 connected as if roasting. Crashed.

Ran Artisan 2.8.2 (the last version I have prior to 2.8.4) connected as if roasting. Did not crash.

Ran Artisan 2.8.5 connected as if roasting. Did not crash but took a reeeeally long time to load and after clicking ‘OFF’ and then closing the window to bring up the Save / Don’t Save dialog box, it never came up and I was never able to bring Artisan back to the front (it’s still open in the background as I type this message). When I click CMD-OPT-ESC, it looks normal and does not show that it’s not responding but it’s not responding.

I know that running a new app for the first time can take a bit long initially and then come up quicker after that but twice now it’s taken a long time for 2.8.5 to come up. I also understand this is not an official release.

My Phidget control panel shows version 102 and there isn’t any indication that there’s a newer version or, as I understand, it would be marked in red and give me the option to click on it to update.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

MAKOMO, 2023-08-21:
> Ran Artisan 2.8.4 NOT connected to the roaster and Phidget - just the app by itself. It did not crash. Ran Artisan 2.8.4 connected as if roasting. Crashed. Ran Artisan 2.8.2 (the last version I have prior to 2.8.4) connected as if roasting. Did not crash. 

Ok. So it seems 2.8.2 is not crashing for you.

> Ran Artisan 2.8.5 connected as if roasting. Did not crash but took a reeeeally long time to load and after clicking ‘OFF’ and then closing the window to bring up the Save / Don’t Save dialog box, it [...]

Hm. Sure on first start apps take somewhat longer. Especially unsigned once. However, I am not experience any longer delays running this on my Intel mac, but for the OFF action. This one takes now exactly one sampling interval to cleanly close down that loop. With a sampling rate set to 1sec which is ok for the fast communicating Phidgets this makes 1sec for the OFF action.

The hang you are experience is indicating that the event loop of the GUI is somehow blocked. This should not happen, and does not happen on any other platform. I fear without access to an M1 mac this cannot be solved.

> My Phidget control panel shows version 102 and there isn’t any indication that there’s a newer version or, as I understand, it would be marked in red and give me the option to click on it to update. 

The Phidget firmware versions are delivered within the Phidget driver not via the network. So if you are not running the latest Phidget driver you are not seeing if your Phidgets firmware is outdated. Thus first install the latest Phidget driver and then check if your Phidget needs an update.

Thanks for testing. Maybe one of the other M1 mac owners can test this as well and report if the behavior is the same as you report it here.

MAKOMO, 2023-08-21:
Dan, maybe you could do one experiment with the that Artisan 2.8.5 beta. Start it, open the Device Assignment menu (`Config >> Devices`) and choose in the first tab as meter not your Phidget (remember which entry it was to set it back after the experiment) and choose `Dummy` from that long list. Dummy is a dummy device reporting always 0 not connecting to any thing. The serial setup dialog that popups up on closing the Device Assignment dialog with Dummy selected can be just closed. 

Now test if app startup and ON/OFF actions are working again as expected. If this is true, the culprit is the Phidget device driver.

fatrabbit-la, 2023-08-22:
Confirming 2.8.2 is not crashing on me.

I reinstalled the Phidget driver even though I think I was already working with the latest one. Restarted the machine just to start fresh. Phidget 22, version 102, July 2023.

Started up 2.8.5. Again, it took a long time to load. I didn’t time it but I’m estimating two minutes.

Changed the device to Dummy and ran/stopped three ‘roasts’ (just running for a few seconds before clicking OFF). Reset the device to my Phidget and ran/stopped a few more ‘roasts’. No issues.

Went back to 2.8.4. Crashed the first time when clicking OFF.

Went back to 2.8.5 and timed how long to load. Force quit after one minute because I just didn’t want to keep waiting.

Went to 2.8.2 and timed it: less than 7 seconds to load.

d

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

CarefreeBuzzBuzz, 2023-08-22:
I just tried with my M1 Mac.  I just installed the latest Phidget driver.  I have RTDs.  I had no crashes.  I started it three times and hit OFF each time with no issues.  Below is my Phidget set up.  

￼

----------------------------------------
Michael Herbert, Head Roaster
Order at www.carefreebuzzbuzz.com <http://www.carefreebuzzbuzz.com/>
Support Artisan RoasterScope  <https://artisan-scope.org/>by using 
Artisan.Plus <https://artisan.plus/en/>

MAKOMO, 2023-08-22:
Dan,

thus the 2.8.5 build is now running not crashing any longer after your Phidgets lib update, but startup is still extrem slow, right? Please attach your settings here. You need to save your settings via menu Help >> Save settings into an artisan-settings.aset file, rename this one to artisan-settings.txt and attach it here on GitHub via drag and drop into a text field (just a reply via email will NOT work!).

What could also help is if you could attach your artisan.log file as well. You can extract this one by holding your OPTION key while clicking on the plus icon in the upper left corner of the Artisan main window. An email should open with all log files attached. Rename all those .log files to read .log.txt and again attach them here on GitHub via drag-and-drop.

I will take a look and see if I can find out what makes your Artisan slow.

Thanks for your support!

fatrabbit-la, 2023-08-22:
I don't think 2.8.5 ever crashed on me the way 2.8.4 consistently crashes as in, it just quits. There was that one instance where 2.8.5 didn't crash but it did hang and I was never able to bring it back to the foreground. I suppose that's a different type of crash but more non-responsive than quitting by itself.

I'm away from the roaster so I hope the files you requested are still relevant. I've attached them here.

Very strange...I opened 2.8.5 (again, not connected to any equipment) in order to export these files and it came up in maybe 10 seconds. I thought perhaps being connected caused the long load time but then I quit and restarted and it did then take a really long time to load.
[artisan_device.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12412980/artisan_device.log.txt)
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12412981/artisan-settings.txt)
[artisan.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12412982/artisan.log.txt)

Dan

MAKOMO, 2023-08-23:
Dan, thanks for testing, this detailed report and for attaching your log files. You seem to start Artisan v2.8.5 directly from its mounted install image. As this is read-only, macOS has to check that whole build for potential security issues on each startup. This might explain the slow start of 2.8.5 for you. Anyhow, I updated the continous build with some more changes that might help to improve the situation for you. Please download the current continuous release build again and this time move the Artisan.app to some location on your internal drive and start it from there via that right-click to the app icon selecting "Open". It should start, at least on second time, faster and without that extra click. Please report back if this resolves your slow-startup issue.

Regarding the hang you observed, please report in detail when it was hanging, so what were the things you did just before the hang. In which situation did that hang occur.

Thanks,
Marko

fatrabbit-la, 2023-08-23:
Marko,

I’ll report back later when I have a chance to test the latest 2.8.5 build but just a note…I never run apps from the install image. I did a full install, deleted the installer, and ran the app locally from my hard drive. And yes, I did have to ctrl-click to authorize it to run.

Dan

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

fatrabbit-la, 2023-08-23:
Ok, I deleted the old 2.8.5 build and installed the most recent version. I deleted the installer - it's being run locally on my computer. It took 2 mins 17 seconds to load which feels about as long as the previous two times. Ran the app (START) for at least 30 seconds (didn't know about this before so thanks for that info) and no issues but I am NOT connected to Phidgets / the roaster. Not sure I'll be able to get to that today.

Regarding the hanging I observed the other day, I honestly don't remember the details. I wasn't expecting it so didn't think to take note of it. I do recall clicking over to check my email or something like that and Artisan was still running, but I was never able to get it back to the foreground. I don't think that has ever happened to me before and not since.
[artisan_device.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12421232/artisan_device.log.txt)
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12421235/artisan-settings.txt)
[artisan.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12421236/artisan.log.txt)

Updated files attached.

MAKOMO, 2023-08-23:
> Ok, I deleted the old 2.8.5 build and installed the most recent version. I deleted the installer - it's being run locally on my computer. It took 2 mins 17 seconds to load which feels about as long as the previous two times. 

According to your attached log, macOS applies App Translocation to Artisan on startup. Thus the whole app is first copied to a random location and then started from there. 

08-23 09:05:20 MainThread artisanlib.main      INFO   : exec: /private/var/folders/0t/wwbt51690nl4c0zmjk57ldd00000gn/T/AppTranslocation/2571E28A-3342-4222-A7DD-C96F9BCFFF26/d/Artisan.app/Contents/MacOS/python

This explains your long startup time, especially if your disk is quite filled and/or your memory situation is tight. App Translocation is a security step that should happen only on the first  run. For you it seems to happen on each run. However, Apple changed the strategy of this measure several times from macOS release to macOS release and is not publishing any details.

https://developer.apple.com/forums/thread/724969

Moving the app to a different folder after first run might help, especially to the standard `/Applications` folder. `App Translocation` seems not to be applied to signed/notarized apps like all regular Artisan releases. However, the details are unclear.

App translations are also reported to cause crashes

https://eclecticlight.co/2022/09/06/dont-run-that-app-where-it-landed-how-translocation-can-cause-crashes/

> Ran the app (START) for at least 30 seconds (didn't know about this before so thanks for that info) and no issues but I am NOT connected to Phidgets / the roaster. Not sure I'll be able to get to that today.

Ok

> Regarding the hanging I observed the other day, I honestly don't remember the details. I wasn't expecting it so didn't think to take note of it. I do recall clicking over to check my email or [...]

Ok. This shouldn't happen and indicates that the main UI thread is busy waiting for something that does not happen. You can only kill Artisan at this point.

[artisan_device.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12421232/artisan_device.log.txt) [artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12421235/artisan-settings.txt) [artisan.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12421236/artisan.log.txt)
> 
> Updated files attached.

I added a signed and notarized version of this latest build here:

https://github.com/artisan-roaster-scope/artisan/releases/tag/m1-test

in the hope this one will show the usual startup speed and the notarization prevents `App Translocation` to happen. If it does I would be interested to see if this one avoids crashes and hangs on regular use or still makes problems on M1 macs.

Thanks for testing!

fatrabbit-la, 2023-08-23:
I have 86GB of available space on my hard drive. I'm currently, and typically, running 8 apps at the same time, none of them memory-intensive: Artisan, Brave browser, Mail, Messages, password manager, calculator, calendar, mail spam app.

I have a dedicated Artisan folder within the Applications folder but went ahead and moved the previous build to just Applications. Took a long time to load. Deleted it and installed the new build. This build seemed to take longer than usual to run the first time but has loaded quickly, as expected, three times in a row.

I ran the app once, NOT connected to the roaster, and it behaved as expected.

Files attached.
[artisan_device.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12423010/artisan_device.log.txt)
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12423011/artisan-settings.txt)
[artisan.log.txt](https://github.com/artisan-roaster-scope/artisan/files/12423012/artisan.log.txt)

fatrabbit-la, 2023-09-01:
I'm happy to report that I just roasted six batches back to back with no crashes or any other bugs. This was with the latest release of 2.8.5.

The only thing that seemed different, but not a big deal, was a big gap added to the end of my graph when I stopped the roast [see attached]. I often tweak this anyway so my graph fits the space perfectly but this gap seems to be a lot bigger than what I recall seeing in the past. But basically a non-issue as far as I'm concerned.

Thanks for addressing the issue for all us M1 users!!!
[image]

closed 2023-10-24 by MAKOMO