artisan-roaster-scope/artisan #1222: Feature Request: Allow event buttons to have a wider margin at bottom of screen

fatrabbit-la, 2023-07-06:
[image]

I have several event buttons at the bottom of my screen. They take up roughly 70% of the overall graph window horizontally before wrapping to the next line. I would like to define the margins on either side in order to add more buttons without them wrapping to the next line.

I've attached an image showing how I want to add a third FAN event but rather than tagging on to the end next to the other fan events, it awkwardly bumps to the next line.

fatrabbit-la, 2023-07-06:
Sorry, I take this back. Not sure why I didn’t see it before but I now see a “Max Buttons Per Row” option which helps me accomplish this.

Ignore.

Fat Rabbit Coffee
***@***.*** ***@***.***>
www.fatrabbitcoffee.com <http://www.fatrabbitcoffee.com/>

roasterdave, 2023-07-06:
Right.  With "Max Buttons per Row" and spacer buttons (buttons that have Visibility=OFF) you can create gaps between buttons and where the next row starts.  From your screenshot it appears you are already using spacers.

closed 2023-07-09 by MAKOMO