artisan-roaster-scope/artisan #1223: Phidgets DCC1000 DCC1003 not working with Artisan

Robjones6146, 2023-07-08:
I am having an issue with the DCC**** Phidgets. They work in the Phidgets control panel. I get -24v and +24v on the multimeter and the linear actuator moves in and out. I can't get the same functionality in Artisan. I am using IO Command with combinations and variations of accel(ch,a[,sn]) sets acceleration (duty cycle / second)
vel(ch,v[,sn]) sets target velocity (duty cycle) limit(ch,v[,sn]) sets current limit

I am using Config >> Events >> Buttons >> Action >>  vel(0,1); accel(0,1) 
I have a button to open the actuator and one to close the actuator, both will be alarms.

I need a voltage output of +24 volts and -24 volts -- same as the Phidgets control panel. I am driving a 24v linear actuator with a 24v PSU connected to the input of the DCC1003 (I had the same issue with the DCC1000) The linear actuator is connected to the output of the DCC1003 channel 0 and the DCC1003 is connected to channel 0 of the Vint hub.

I am on Windows 10
Artisan 2.8.2 (f525b16)
Phidgets Vint Hub
Phidgets  DCC1003
24v PSU
24v linear actuator

Thank you in advance.

MAKOMO, 2023-07-08:
First, please update to the current Artisan v2.8.4. We do not have the resources to support outdated versions.

Next, activate "Logging" in the Device Assignment dialog (menu Config >> Device). Hold ALT and CTRL while you click on the plus icon in the upper left corner to toggle Artisans logging to DEBUG (observe the message line until it reports "debug logging ON". Now connect your hardware and press your buttons.

Finally, send us your Artisan log files by holding ALT and clicking again on the plus icon. An email should open with the log files attached. Send them either to info@artisan.plus or rename them here to end with .txt and attache them here on Github by drag-and-drop them to a answer text field. Please also add your Artisan settings file artisan-settings.aset (also renamed to end in .txt). You can export this via menu Help >> Export Settings.

We will take a review your logs and settings as soon as possible.

Robjones6146, 2023-07-08:
I have upgraded. Thank you. Let me know if what I emailed is not correct or sufficient.

MAKOMO, 2023-07-08:
I don't know much about those motor controllers and don't have one at hand. I assume you are using the two buttons:

1. To\nTub: IO Command "accel(0,1,:0)"
2. To\nPack: IO Command "accel(0,-1,:0)"

to set the Acceleration to -1 or 1.

In the artisan.log I see an error message from the Phidgets lib:

```
07-08 09:53:50 Dummy-5    artisanlib.comm      ERROR  : PhidgetException 0x15 (Invalid Argument)
Value must be in range: 0.100000 - 100.000000.
Traceback (most recent call last):
  File "artisanlib\comm.py", line 4510, in phidgetDCMotorSetAcceleration
  File "Phidget22\Devices\DCMotor.py", line 132, in setAcceleration
````

According to the [Phidget documentation](https://www.phidgets.com/?tier=3&catid=18&pcid=15&prodid=1118) acceleration can only be set to values in the range [0.1, 100].

Maybe you intended to set the Velocity instead? For this you need to use the IO Commands "vel(0,1,:0)", "vel(0,-1,:0), respectively"?

There are other things in your settings that look strange to me. Your Hot air slider uses "accel" and "vel" commands in an `Artisan Command` action which does not support DCMotor commands according to its definition (see the Help page) accessible via the HELP button in the Buttons tab.

You configured an extra device of type "DAQ1400 Frequency" but according to your device.log you never attached one. Even worse, you added another extra device of type "DAQ1400". Then you have an extra device of type "TMP1101 4xTC23" to access the TCs connected to inputs 2+3 of your TMP1101. Fine. But why adding another extra device of the same type. Note that the first TMP1101 4xTC23 already reads both channels, 2 and 3. Then you also added another extra device of type "HUB IO 0" This will attach to that HUB0000 port 0 and block it, but it seems to be unused!?

Robjones6146, 2023-07-08:
Hi,

As I got your email I was halfway to getting there. I got +24v and now
changing the Vel to minus and not the accel (like in the control panel) It
works!!!

Thank you so much!!

I designed the system before I got the Phidgets. There is quite a lot going
on.

4 x thermocouples (BT, ET, Cyclone abd Air in)
3 x linear actuators (beans in/out and scale hopper)
2 x Belimo damper actuators with voltage feedback
2 x pressure sensors (roast chamber and cyclone afterburner)
2 x relays (gas burner and chaff auger motor)
1 x PWM (to control a VFD 3-phase blower motor)
1 x solenoid (water quench)

I am working my way through them all slowly, learning and trying to get
them to work.

The extra devices are for blower speed, air pressure, and 2 extra
thermocouples for Cyclone afterburner temperature and air in temperature.
That's the plan. I got stuck on the DCC1000 for hours, going around in
circles.

I am designing a fluid bed roaster for my brother in the UK. When it is up
and running I have requested he make a donation.

Thank you for your fantastic software and your help today.
I am sure I will need more help and advice in the weeks to come.

regards

Robin

closed 2023-07-08 by MAKOMO