artisan-roaster-scope/artisan #1224: Extra Devices - Vint Hub >> DAQ1000 > Phidgets1136 (pressure sensor) No support DAQ1000   8 x voltage input

Robjones6146, 2023-07-09:
Might not be a bug or a request. I might be missing something.

I am trying to connect 2 x pressure sensors. One Phidget 1136 and one generic. Both Voltage/analog. They are both connected to the 8 x voltage input Phidgets DAQ1000. There is no option to select this from the dropdown in Extra devices and it is also not listed on the Phidgets layout Device Assignment >> Phidgets

https://www.phidgets.com/?tier=3&catid=7&pcid=5&prodid=109
https://www.phidgets.com/?tier=3&catid=106&pcid=86&prodid=622

Is it possible to use another option from the dropdown?

I have tried a few combinations:
Phidgets HUB IO 23 (Vint HUB port 3 and DAQ1000 port 2/3)
Phidgets IO 23
Phidgets HUB IO Digital 23

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/11995412/artisan-settings.txt)

I am on Windows 10
Artisan 2.8.4 (ac17fbe) 
Phidgets Vint Hub
Many Phidgets

Thank you.

MAKOMO, 2023-07-09:
> I am trying to connect 2 x pressure sensors. One Phidget 1136 and one generic. Both Voltage/analog. They are both connected to the 8 x voltage input Phidgets DAQ1000. There is no option to select [...]

The DAQ1000 is not listed among the [Phidget modules supported by Artisan](https://artisan-scope.org/devices/phidgets/)

> Is it possible to use another option from the dropdown?
> 
> I have tried a few combinations: Phidgets HUB IO 23 (Vint HUB port 3 and DAQ1000 port 2/3) Phidgets IO 23 Phidgets HUB IO Digital 23

No

> 
> [artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/11995412/artisan-settings.txt)
> 
> I am on Windows 10 Artisan 2.8.4 ([ac17fbe](https://github.com/artisan-roaster-scope/artisan/commit/ac17fbefe5e140abb1dc604a05f791cc284d8dd3)) Phidgets Vint Hub Many Phidgets
> 
> Thank you.

Voltage Input is currently only supported via the HUB ports (`Phidget HUB IO`) or the DAQ1400 (`DAQ1400 Voltage`).

PRs are welcome!

Robjones6146, 2023-07-09:
Hi Makomo,

Should I request a DAQ1000? It seems very versatile as I understand it will take any device that outputs a voltage between 0 - 5v. It also has 8 inputs! I also want to get a frequency or RPM of my blower motor/VFD/PWM, a voltage should work.

Robin

Robjones6146, 2023-07-09:
Hi Makomo,

Is this a feature I need to request? Whats a PR?

Robin

MAKOMO, 2023-07-09:
Hi Robin,

this is an open-source project. Sure you can request features, but there is no guarantee requested features will be implemented at any time. A PR is the abbreviation for a `Pull Request`, a mechanism how one can contribute code to this project implementing features, changes and improvements.

Robjones6146, 2023-07-09:
Hi Marko,

I understand and I appreciate all the work that everyone has done on this fantastic software. 

I see that the1136 is listed as supported, maybe I am missing something. There must be another way for Artisan to access the sensor that I don't know about.

I might turn to one of the blogs for suggestions. I am sure there is someone that has a pressure sensor attached to their roaster.

I will add a feature request for the DAQ1000 in the meantime. I think it will be a valuable addition for extending the No. of inputs when using a Vint Hub, with all the available analog sensors out there.

Thanks for your assistance.

Robin

MAKOMO, 2023-07-09:
> Hi Marko,
> 
> I understand and I appreciate all the work that everyone has done on this fantastic software.
> 
> I see that the1136 is listed as supported, maybe I am missing something. There must be another way for Artisan to access the sensor that I don't know about.

Yep, connect it directly to a HUB port and use the HUB IO device types.

> 
> I might turn to one of the blogs for suggestions. I am sure there is someone that has a pressure sensor attached to their roaster.

Yes, many. Myself included.

> 
> I will add a feature request for the DAQ1000 in the meantime. I think it will be a valuable addition for extending the No. of inputs when using a Vint Hub, with all the available analog sensors out there.

Sure

> 
> Thanks for your assistance.
> 
> Robin

closed 2023-07-10 by MAKOMO