artisan-roaster-scope/artisan #1225: Support for Phidgets DAQ1000 eight channel analog (voltage) input

Robjones6146, 2023-07-09:
I would like to request the addition of support for the Phidgets DAQ1000 eight-channel voltage input.
https://www.phidgets.com/?tier=3&catid=106&pcid=86&prodid=622

I want to capture the air pressure of my roast chamber and cyclone afterburner of a fluid bed roaster. I am using a Phidgets 1136 pressure sensor and another generic pressure sensor both using 5 volts. The voltage inputs will extend the number of analog inputs when using the Vint Hub. With the amount of analog sensors available, the DAQ1000 will add 8 additional channels.

I also want to capture the RPM of the blower possibly through the VFD or PWM voltage so I can monitor the loft required for the batch size.

It might be similar to the 1018. This is not suitable for me as I am using the Vint Hub. I have tried the DAQ1400, but there is an issue with the 12/24 volts, the 1136 uses 5 volts.

Thank you in advance.

Robin

MAKOMO, 2023-07-09:
You are right. It looks similar to the 1018. Maybe a small extension could make this happen. I will take a look once I find the time.

Robjones6146, 2023-07-10:
Thank you Marko. 
May I request a factor or calibration field? The DAQ1400 also has a voltage input and might be similar.

Robjones6146, 2023-07-10:
It looks like you created the update. What do I need to do to get the update or how?

MAKOMO, 2023-07-10:
I added some code to support the DAQ1000. This is untested as I don't have access to this module. There are new device types `Phidget DAQ1000 01`, `Phidget DAQ1000 23`, `Phidget DAQ1000 45` and `Phidget DAQ1000 67`. They follow the configuration of the Phidget 1018 module as configured in the Phidgets tab of the Device Assignment dialog (menu `Config >> Device`). So you can decide here per channel for example if they should work in async mode or in voltage ratio mode, among others.

A factor and/or calibration can be applied as for any Artisan channel using a corresponding [Symbolic Formula](https://artisan-roasterscope.blogspot.com/2019/11/symbolic-formulas-basics-new-variables.html).

This addition is now part of the [Continuous Builds](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous).

Please give this a try and report back if you get the expected attach message and maybe even some data from your device.

Robjones6146, 2023-07-10:
Fantastic! Thank you so much. That works perfectly. I got pressure in the LCD as well as a pseudo motor speed too when I send the voltage from the OUT1002 (pwm for my VFD to blower) to the DAQ1000. The symbolic formula is very useful, thanks for the tip.

Good job and so quick. 
Thanks, Marko.

closed 2023-07-10 by MAKOMO

darrenaddy, 2023-07-11:
> ID: <artisan-roaster-scope/artisan/issue/1225/issue_event/9772861671@
> github.com>

Impressive Marko!
Assuming you updated just the src/artisanlib/phidgets.py ?
Can one simply replace that file or does one need to download the whole
latest continuous build?

Darren Addy
Kearney, NE

MAKOMO, 2023-07-11:
Several files got changes as you see from my commit. You need to download the whole build.