artisan-roaster-scope/artisan #1230: Send the opposite value of a slider controlled by Artisan PID to another slider

Robjones6146, 2023-07-13:
I would like to send the opposite value of a slider to another slider that is controlled by Artisan PID. When the Hot air slider is at 60 (6 volts output), I want the Cold air slider to be 40 (4 volts) and visa versa.

I have two damper paddles that need to equal 100% (if the cold is 20% then the hot must be 80% and visa versa) I am using the Set Value >> Background in PID Control.  I am using two OUT1002 Phidgets that send 0 - 10v signals to two Belimo rotary actuators that rotate the damper paddle through 90 degrees (The Belimo works with a 0 - 10 volt signal for positioning). The voltage corresponds nicely with the damper position eg. 3 volts = 30% open. I want to send the other actuator 7 volts (10 - 3 = 7) for 70% open.

I have tried achieving this in PID Control with no success. (Lots of variations)
PID settings (p - 3, i - 0.05, d - 10)
Positive - Cold air
Negative - Hot air
Invert Control - checked
Duty Max - 100%
Duty Min -100%

Windows 10
Artisan 2.8.5 (47c8ae6)
Phidgets Vint Hub

I have requested assistance in the Discussions where there is a detailed description. [Artisan PID question. Control two damper paddles controlled via sliders](https://github.com/artisan-roaster-scope/artisan/discussions/1227)

I can't find anything in the documentation that deals with this issue.
I would appreciate any hardware or software advice or solution to achieve this.

Robjones6146, 2023-07-16:
Can anybody help me with this please?

MAKOMO, 2023-08-28:
What you request is simply not directly possible. However, you can go with just one slider that sends two commands (separated by a semicolon), the first just sends the slider value v and the other the 100-v.

closed 2023-08-28 by MAKOMO

Robjones6146, 2024-10-27:
I know this is over a year ago. I am only implementing it now and cant get it to work.

Please can you confirm that this is correct. In the Sliders Tab:
Event (Slider): VOUT Command  out(0,{},687340:4);out(0,100-{},687340:3) 

- No space after semicolon
- replace v with {}
- 
[image]

MAKOMO, 2024-10-27:
You forgot to state WHAT exactly is not working. Your slider VOUT command looks fine.

Robjones6146, 2024-10-27:
Thank you, I am sure it will work now.

 I haven't tested this  out(0,{},687340:4);out(0,100-{},687340:3)  yet, I
only have access to the machine on Monday.

I tried various combination:
out(0,{},687340:4); out(0,100-v,687340:3)

- I had a space after the semicolon
- I used your suggestion literally

Using this Slider for PID
Using the the two slider method, the Hot Slider has to reach 0% before the
Cold Slider starts moving (+100% to -100%)
This way when Hot is 80% then Cold is 20%