artisan-roaster-scope/artisan #1235: Artisan 2.8.4 crashing on roast end / clicking OFF

cbuehling, 2023-07-21:
**Describe the bug**
Artisan 2.8.4 crashes on pressing OFF after roast end; happened ~3 out of the past 5 times.

**To Reproduce**
Steps to reproduce the behavior:
1. Run your roast as normal
2. Click on 'OFF'
3. Artisan crashes, see stack trace attached
4. 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Setup (please complete the following information):**
 - Artisan Version [e.g. v2.4.4]: 2.8.4
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]: ac17fbe
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]: 13.4.1 (22F82)
 - Connected devices or roasting machine [e.g. Probatone 5]: Hive

**Additional context**
github won't let me attach the settings.aset

_Note that you nee
[artisan_crash.txt](https://github.com/artisan-roaster-scope/artisan/files/12135137/artisan_crash.txt)

MAKOMO, 2023-07-22:
This is on an M1 mac? Seems related to https://github.com/artisan-roaster-scope/artisan/issues/1221
There seems to be a conflict between the Qt Gui Tool Artisan is using and MacOS on M1 macs for Artisan v2.8.4. Do you see the same crasher using Artisan 2.8.2?

You can attach documents per drag-and-drop to GitHub only documents with extension .txt. Just rename to settings.txt

cbuehling, 2023-07-24:
Thanks for looking into it so quickly. Yes, it's an Apple M1 Max.

I don't think I ever saw a crash on any previous version. Here's another crash from today and the settings file renamed to txt

[artisan_crash2.txt](https://github.com/artisan-roaster-scope/artisan/files/12141514/artisan_crash_2.txt)
[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12141516/artisan-settings.txt)

MAKOMO, 2023-08-23:
Please follow the Issue #1221 which reports similar issues on M1 macs. Note that an updated [Continuous build](https://github.com/artisan-roaster-scope/artisan/releases/tag/continuous) has been released that might resolve your issue. Please report back your test results along that other issue. Thanks!

closed 2023-08-23 by MAKOMO