artisan-roaster-scope/artisan #1236: An unhandled exception occurred

xxiphonexx, 2023-07-22:
The app stopped and showed this message 
"An unhandled exception occurred. Please report the problem on Github:
https://github.com/artisan-roaster-scope/artisan/issues

When reporting this issue, please include your settings file (export 
via menu Help >> Save Settings) and the details below.

An entry has been written to the error log (menu Help >> Error).

Version: 2.8.4, revision: ac17fbe
2023-07-22, 01:37:05 -------------------------------------------------------------------------------- : list index out of range (line: 22912)"

Error Log:
"version = 2.8.4

3 01:37:05.044 Error: File "artisanlib\main.py", line 22912, in background File "artisanlib\background.py", line 167, in __init__ File "artisanlib\background.py", line 792, in createEventTable -------------------------------------------------------------------------------- artisanlib\main.py::background:22912 self _False backgroundDlg artisanlib\background.py::__init__:167 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: {3}', 'B4: {0}', 'B5: SV', 'B6: '] i1 __class__ artisanlib\background.py::createEventTable:792 self ndata8 start0 i5 timez fmtstr%.1f etline btline description etype evalue 

2 01:37:02.018 Error: File "artisanlib\main.py", line 22912, in background File "artisanlib\background.py", line 167, in __init__ File "artisanlib\background.py", line 792, in createEventTable -------------------------------------------------------------------------------- artisanlib\main.py::background:22912 self _False backgroundDlg artisanlib\background.py::__init__:167 self parent aw activeTab0 settings backgroundDeltaETflagLabel backgroundDeltaBTflagLabel loadButton delButton alignButton curvenames['', 'B3: {3}', 'B4: {0}', 'B5: SV', 'B6: '] i1 __class__ artisanlib\background.py::createEventTable:792 self ndata8 start0 i5 timez fmtstr%.1f etline btline description etype evalue 

1 01:36:58.414 Exception: redraw() list index out of range@line 8383

"

[artisan-settings.txt](https://github.com/artisan-roaster-scope/artisan/files/12136651/artisan-settings.txt)

MAKOMO, 2023-07-22:
We need something to reproduce this. It seems you opened the Background Dialog when you observed this exception, right?

Could you please also attach the Artisan profile as .alog file (renamed to end with .txt) here such that we can try to reproduce this?

Thanks for your support!

xxiphonexx, 2023-07-22:
Hi Marko,
Yes I was using Background.
I got the error yesterday and do not remember which one it is and will upload two profile.  Not sure if it is related but both have Arabic letters in them and it did not make an error before. 
[عربي 10auto_23-07-22_0125.txt](https://github.com/artisan-roaster-scope/artisan/files/12137145/10auto_23-07-22_0125.txt)
[عربي 11auto_23-07-22_0125.txt](https://github.com/artisan-roaster-scope/artisan/files/12137146/11auto_23-07-22_0125.txt)
Thanks you for your help

roasterdave, 2023-07-24:
Thanks for reporting.  Can you provide any information about the source of these profiles?  They do not look like they came directly from recording a roast.  Most likely they are output from the Designer after reading other profiles into the Designer.  Were there any edits outside of Artisan?

The unhandled exception is caused by inconsistent data in 10auto_23-07-22_0125.alog.  This inconsistency has not not been seen  or reported before.  Any additional information will help to identify how the unexpected data was introduced, so if there is a bug we can find and fix it.

xxiphonexx, 2023-08-04:
Hi Dave,

Sorry for the late reply, hope you are doing well.

I do not have any information because this is my start with Artisan. I did
not notice anything and unfortunately I could not find
10auto_23-07-22_0125.alog. As for you questions yes I was creating
something new in Designer and the did not do edits outside of Artisan.

roasterdave, 2023-08-04:
Ok.  Thanks for that information.

We are not able reproduce this.  Some protective code will be added that will prevent the unhandled exception.  However, the  root cause of the inconsistent data in 10auto_23-07-22_0125.alog is unknown.

MAKOMO, 2023-11-14:
Added internal event API as countermeasure to such issues (Commit e44de8a1a70b6691170bf42fc16000024c5fc9c6)

closed 2023-11-14 by MAKOMO