artisan-roaster-scope/artisan #1248: Artisan on ARM (aarch) Arch Linux

dvizard, 2023-08-07:
**Is your feature request related to a problem? Please describe.**

There is an AUR package for Artisan at https://aur.archlinux.org/packages/artisan-roaster-scope. This somehow downloads the x64 debian package and installs that. 

There is also a Raspberry Pi package for armhf, suggesting that x64 is not essential to the software.

I am trying to get Artisan to run on an aarch64 device running Arch ARM (the PineTab 2, https://wiki.pine64.org/wiki/PineTab2). 

Attempt 1:
* clone AUR
* modify architecture to aarch64
* modify version to 2.8.2
* modify source deb to raspi deb from release 2.8.2
* fix the SHA256 sum
* install with makepkg -si

The installation works, but nothing happens when I try to start Artisan. Since the runner script sends errors to dev/null, I don't know yet what's going on.

Attempt 2: 
I have tried following the ["run from source" instructions](https://artisan-roasterscope.blogspot.com/2019/02/running-artisan-on-raspberry-pi.html) for Raspberry Pi and adapting to arch linux where necessary, but fail while `pip install -r requirements.txt`  during the step: https://gist.github.com/dvizard/db191a5292107c5c83d2a3609d5340e8. On top, this might not be viable at all: I can install `libphidget22` drivers from AUR but have not yet found any `libphidget22extras`  or `phidget22networkserver`. I don't need phidget, but from the doc it seems that this is a requirement for running from source.

I'm surprised to see pyqt6 errors when the instructions say to download pyt5 tools, is this possibly outdated?

**Describe the solution you'd like**
Any way of getting it to work :D

**Additional context**
I am running a selfbuilt arduino thing and want to use my pinetab2 as the frontend for it (instead of bringing my full laptop). Using Artisan on Windows, everything works already.

dvizard, 2023-08-19:
Got Artisan to work on Mobian (Debian) arm64.
The main issue is that there is no pipy wheel for PyQt6, and building it is a massive PITA. I even managed to build it by hand but for some reason that version still didn't work. Solution: use prebuilt PyQt6 from system packages. Phidget drivers also need to be built, but that's easy.

Steps:
* Build Phidget, extra, network from source
* Install pyqt6 from system package manager
* Create a python virtualenv with `--system-site-packages` so it gets the system pyqt6
* Clone Artisan from Git
* Remove manually from requirements.txt:
  * PyQt6
  * PyQt6-webengine (since the code seems to say this is optional)
  * For good measure, also the PyQt5 and webengine
  * qt5-tools
  * qt6-tools
* `pip install -r requirements.txt`

wvengen, 2024-08-09:
Note that Artisan has just been [released on Flathub](https://flathub.org/apps/org.artisan_scope.artisan), which includes a package for `aarch64`. This may be the easiest way to get going (see [setting up Flatpak](https://flathub.org/setup/Arch) if you haven't used Flatpak before).

closed 2024-08-15 by MAKOMO