artisan-roaster-scope/artisan #1251: Feature request. Add support for HUM1000 Phidget to display in LCD and chart.

Robjones6146, 2023-08-15:
May I request support for the HUM1000 to display on the LCD. I am going to be water-quenching my roast to cool it quickly. My beans cool in the roast chamber.

It would be great to get some consistency with the amount of water sprayed over the beans. 
The HUM1000 is supported as an ambient temperature and humidity Phidget. I hope that it won't take too much effort to support this as a general Phidget.

I wanted to connect a serial humidity sensor FS200-SHT10 using an Arduino MEGA 2560 (USB) via Artisan MODBUS (serial). I added this to the Discussions a few weeks back. I will keep pursuing this in the future as it is a much more robust solution considering the high temperatures.

I am sure that home roasters could use this feature. I will be interested to see the graph around first crack when steam is released. Although this is not my main interest it could lead to some interesting comparatives of different beans and the moisture content and weight loss.
[image]

Thank you Marko in advance. I hope this is something you can undertake.

MAKOMO, 2023-08-20:
Rejected. This application is out of the specification of this module. This module is fully supported for what it is intended for. Measuring of ambient temperature and humidity in standard ambient limits.

closed 2023-08-20 by MAKOMO

Robjones6146, 2023-08-20:
Thanks for your reply. I will continue to pursue using the FS200-SHT10 with an Arduino MEGA 2560 (USB) via Artisan MODBUS serial.

MAKOMO, 2026-01-22:
Support for the Phidget TMP1000, HUM1000 and PRE1000 as extra devices added for the next Artisan release