artisan-roaster-scope/artisan #1253: Data format change in .alog file??

hazardtrip, 2023-08-21:
## Expected Behavior
I have a spreadsheet running power queries off the .alog files to extract the data from artisan and assist with reporting and QA,  This functioned perfectly until the last upgrade from 2.6.0 to 2.8.4.  Now it appears the the field and value data has swapped?  I'm trying to troubleshoot this but really need to know what has changed in the data format of the .alog files since 2.6.0  I've reviewed the change notes but didn't see anything there, perhaps I missed something? 

## Actual Behavior
Report from .alog files no longer function as expected since upgrade

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Specifications

  - Artisan Version: 2.8.4
  - Artisan Build (number in brackets shown in the about box): ac17fbe
  - Platform (Mac/Windows/Linux + OS version): Windows 10
  - Connected devices or roasting machine: Diedrich 6 probe

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[using version 2_8_0 1659_23-08-15_1808.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/12402823/using.version.2_8_0.1659_23-08-15_1808.alog.txt)
[using version 2_6_0 1518_23-06-16_2108.alog.txt](https://github.com/artisan-roaster-scope/artisan/files/12402825/using.version.2_6_0.1518_23-06-16_2108.alog.txt)

[artisan-settings 22082023.aset.txt](https://github.com/artisan-roaster-scope/artisan/files/12402827/artisan-settings.22082023.aset.txt)

roasterdave, 2023-08-21:
What specifically does not work?  Do your queries parse the file or rely on position in the file?  There have been no data format changes and certainly no field and value swaps.  Some new fields were added but they will not interfere with a fully qualified key:value query.

hazardtrip, 2023-08-24:
OK, thanks for the reply, I've narrowed it down...

Theres a duplication of data in the alog files on the new version.  Right at the start it adds 'recording_version': '2.8.4', 'recording_revision': 'ac17fbe', 'recording_build': '0' in addition to the existing data.  This is causing my power query to read the value/data incorrectly.  I can set it to ignore the first bit.  I'm wondering why this was added, just curious?

{**'recording_version': '2.8.4', 'recording_revision': 'ac17fbe', 'recording_build': '0',** 'version': '2.8.4', 'revision': 'ac17fbe', 'build': '0'

roasterdave, 2023-08-25:
Good that you found the problem.  Do your queries rely on position rather than fully qualifying the key field?  

This is not duplication.  These fields have a quite different meaning.  Every time Artisan saves a profile information about the currently active Artisan instance is stored in the 'version', 'revision' and 'build' fields.  These fields will be changed if a profile is edited in a different Artisan version.  Version 2.8.2 added 'recording_version', 'recording_revision', and 'recording_build' fields that are written once and only by the Artisan instance that created the profile and are not changed if the profile is later modified by another version of Artisan.

[image]

hazardtrip, 2023-09-25:
Hey, sorry for slow reply, turns out we were just a little too specific in our logic, loosened it up and we are now getting all the data, thanks for your help.

closed 2023-09-25 by MAKOMO