artisan-roaster-scope/artisan #1256: above&below of alarms/condition are reversed in Japanese language

hiroyuki2080, 2023-08-24:
**Describe the bug**
above&below of alarms/condition are reversed in Japanese language

**To Reproduce**
Steps to reproduce the behavior:
at any time

**Expected behavior**
the reversed expressions in Japanese  should be fixed

**Screenshots**
added

**Setup (please complete the following information):**
 - Artisan Version 2.8.2
 - Artisan Build (number in brackets shown in the about box) [e.g. 97e6dec]:f525b16
 - Platform (Mac/Windows/Linux + OS version) [e.g. macOS 10.15.3]:Windows10
 - Connected devices or roasting machine [e.g. Probatone 5]:TC4
[image]

**Additional context**
Add any other context about the problem here.

Please attach your current Artisan settings file (as exported via menu Help >> Save Settings as *.aset) file.
Please attach any relevant Artisan *.alog profiles.

_Note that you need either add a `.txt` extension or zip the files before uploading. Otherwise you will receive a "Not a supported file type" error on uploading._
[image]

MAKOMO, 2023-08-28:
Thank you very much for reporting this issue and for your very clear and detailed report. I just fixed the translations and the update will be available in the next Artisan release.

If you find any there other issues with those translations, please report and we will fix them immediately.

closed 2023-08-28 by MAKOMO

hiroyuki2080, 2023-08-29:
Thank you very much for your quick action.